scikit-rf/scikit-rf #47: network connect more than one consecutive port is wrong

Waterkeyn, 2015-02-05:
In network.connect and same functions, all the connected ports are removed at first pass although only one connection has been made and iteration is called just after.
The lines 2572 to 2574 (scikit-rf 0.14) 
    # combine z0 arrays and remove ports which were `connected`
    ntwkC.z0 = npy.hstack(
        (npy.delete(ntwkA.z0, range(k,k+num), 1), npy.delete(ntwkB.z0, range(l,l+num), 1)))
should become e.g.
    # combine z0 arrays and remove ports which were `connected`
    ntwkC.z0 = npy.hstack(
        (npy.delete(ntwkA.z0, range(k,k+1), 1), npy.delete(ntwkB.z0, range(l,l+1), 1)))
only deleting one port (code optimization to be done)

Thank you for scikit-rf, very useful!

arsenovic, 2015-02-05:
im sorry but i dont quite understand the problem you are describing. could you post an example which illustrates the failure.

arsenovic, 2015-02-05:
ok, i see what your fix is adressing. 

 i think this illustrates that it is working now. i have also added a test, but it only verifies hte z0 order, not the s-paramter accuracy

https://gist.github.com/a178b56251915ac8528c

arsenovic, 2015-02-20:
@Waterkeyn  

is this issue resolved?

arsenovic, 2015-05-26:
ill assume this issue is fixed,  and close this

closed 2015-05-26 by arsenovic