scikit-rf/scikit-rf #49: Pyc files under version control

TheWylieStCoyote, 2015-02-09:
I am not sure why there are .pyc under version control in the [skrf/pythonicsInstruments](https://github.com/scikit-rf/scikit-rf/tree/master/skrf/pythicsInstruments) folder. Moreover the source files are not  present.

arsenovic, 2015-02-10:
yes i probably meant to remove these. i ll do this. 

if you are interested in pythics, you probably can find the files in the git history. 

i have moved to using either 
*  the ipython notebook's  widgets for simple ui's
* traitsui for more complex UI's

arsenovic, 2015-02-10:
[here](http://nbviewer.ipython.org/github/scikit-rf/examples/blob/master/VNA%20Control.ipynb) is an example of simple VNA data retrieval with ipython's interact function. 

also note that it uses the new python-ivi backend, which is a very recent addition

arsenovic, 2016-02-08:
i removed that folder, as it was really old and not supported

closed 2016-02-08 by arsenovic