scikit-rf/scikit-rf #56: Wrong ylabel in NetworkSet.plot_uncertainty_bounds_s_db()

core-dd, 2015-05-19:
Hi!
Unfortunatly, the ylabel generated by NetworkSet.plot_uncertainty_bounds_s_db() is "Magnitude" instead of "Magnitude (dB)". 
Most likely the overwritten attribute "s_mag" in line https://github.com/scikit-rf/scikit-rf/blob/master/skrf/networkSet.py#L910 is the reason. Have no idea how to fix that quickly.

arsenovic, 2015-05-19:
the problem here is that the statistics are run on the `s_mag`, but need to be converted to log scale (s_db), when plotted. to get around this i used the `ppf` as a argument to the function. you could simply add a manual `ylabel` call directly in `NetworkSet.plot_uncertainty_bounds_s_db()` ?

the root of the problem is that the design is bad..

core-dd, 2015-05-20:
Ok. We can do that the quick and dirty way... ;) Maybe I manage to write that quick fix today

arsenovic, 2015-05-26:
cool ill close this.

closed 2015-05-26 by arsenovic