scikit-rf/scikit-rf #57: Generalized mixed-mode S-parameters

core-dd, 2015-05-20:
I saw that you started implementing conversion functions between single-ended and mixed-mode S-parameters. How far have you come and are you going to follow up on that enhancement?

I've stumbled over ["Generalized Mixed-Mode S-Parameters"](http://dx.doi.org/10.1109/TMTT.2005.860497) and consider to implement that in the not to far future.

Any thoughts?

arsenovic, 2015-05-20:
yes, the `mixed mode` branch (https://github.com/scikit-rf/scikit-rf/tree/mixed-mode)  fell by the wayside. im not sure why i stopped working on this, but i dont use this feature, so its not suprising that it fell through the cracks. either way,  i would be happy to help with its development. if you would like any help, or want to discuss the implementation, let me know.

core-dd, 2015-12-18:
First implementation done by request #65. Solution is working at the moment :)

closed 2015-12-18 by core-dd