scikit-rf/scikit-rf #67: Time Domain Representations

core-dd, 2015-12-18:
Hej hej,
is there any interest to discuss transformations into time domain to approach device behavior from a different perspective. There are many possible ideas on which transform (FFTs, z-Transforms, etc) works better. Even some time domain gating could be interesting to apply to measured real world data. What are your thoughts?

Cheers,
Jens

arsenovic, 2015-12-18:
i would be very interested in this. i have worked on  the time gating  with mixed success.  currently, the phase information is not preserved corrected. i have yet to find an explicit explanation of time gating that made sense too me, but i would really like to get this feature working, as a lot of people could make use of it. 

alternative time-domain algorithms can only help.

arsenovic, 2016-02-08:
@core-dd , 
     icurious if you are still interested in working on this? i am about to take another stab at.

core-dd, 2016-02-08:
Still interested in that feature. Had a go with a VNA integrated TD solution and some indoor antenna measurements a while ago. Nice to figure out which reflections are due to the roof and walls and applying some time gating to the result. It's very busy at work right now. Hence, no real time to dig into that feature on my side right now.

arsenovic, 2016-03-05:
i have updated the time-gate algorithm to support center/span inputs, as well as bandstop vs bandpass mode.  
https://github.com/scikit-rf/scikit-rf/blob/master/skrf/network.py#L2413

if you have some antenna  measurements you could test this on, i would be interested to see the results. mostly i have applied the gating in rectangular waveguide.

core-dd, 2016-05-12:
Hej Alex,
I've had a quick look on the time domain representation for some cable measurements over here. The result looks very promising. I'll get back to you once I've made some antenna measurements in the vicinity of some reflectors with the proper frequency domain settings to get enough time domain resolution. I'll share the results here. Thank you for the excellent work!

core-dd, 2016-05-12:
Did some measurements this afternoon and post-processed the the scattering parameters with skrf. The setup consisted of one of our antennas plus a 4.5 m long cable (considered as part of the DUT and calibrated out for comparison).

[image]
This figure compares the TD response with and without cable. The additional time delay is clearly visible and the 4.5 m cable length relate well to it (reflection length and velocity factor).

[image]
I played around with the time gating to see, whether I can tackle the reflections on the cable. Further investigation of applying the right filters for the right time span might lead to comparable results. The plot shows, that the fast ripple vanishes by applying a time gate to the main response of the antenna.

[image]
As last experiment, I placed a metal sheet about 19 cm above the antenna. The plot shows that both TD responses are the same up to about 5 ns, afterwards they diverge over time as the metal plate contributes additional reflections. Unfortunately, I can't make out a particular reflection of the plate (expectation of a bit more then one nanosecond delay w.r.t. to the main antenna reflection). I would assume that the signature is buried by other contributions of the antenna structure.

Your implementation does a nice job! It might be interesting to allow an additional x-axis accounting for free space distance (divided by two).

arsenovic, 2016-05-13:
glad you can get some use out of it. 
are you using skrf for calibration?

also,  gating is only accurate for the first reflection. ie, if the response of interest is behind a big reflection, gating will be inaccurate. the inaccuracy is  in proportion to the size of the first reflection.  this is due to wave mechanics, not the algorithm. 

if you want to add pics of the setup and some code, we could make a 'example' notebook for skrf docs. up to you.

arsenovic, 2016-05-13:
also you can could make a model for your cable (using [skrf.media.coax](http://scikit-rf.readthedocs.io/en/latest/reference/media/generated/skrf.media.coaxial.Coaxial.html#skrf.media.coaxial.Coaxial)) and de-embed a section of it (`coax.line(4.5,'m').inv**dut`). this is not as good as just calibrating it out, but sometimes its  the only way. 

you can also just measure the cable by itself, and save it as a network for de-embedding. im sure you know this already.

core-dd, 2016-05-16:
I'm not using skrf for calibration issues right now. We are trying to get a hold on noise figure measurements for some other sub-project and there it might be handy to look deeper into it.
Thank you for elaborating the gating accuracy behind big reflections. Haven't put much thought into it, but it's clear that it will have a strong influence as the "pulse shape" is altered already.

jhillairet, 2020-07-26:
Some time domain functions have been added since. Not sure if it solves the need. Please reopen if necessary.

closed 2020-07-26 by jhillairet