scikit-rf/scikit-rf #73: Acknowledgement/Citation of skrf?

core-dd, 2016-05-16:
Dear all,
as there was a lot of time invested into skrf over the last years, I'm wondering on how to acknowledge the beautiful work of all involved in an efficient manner in a scientific context. E.g. the Astropy community is going a way documented at [Astropy: About](http://www.astropy.org/about.html#acknowledge). Are there any opinions?

Cheers,
Jens

arsenovic, 2016-05-17:
hey jens, 
   we dont have any technical article to cite for skrf (although andrew temme @temmeand  does have a draft for one which was never published.) 

i think adding  a section to the website indicated how to acknowledge skrf (like astropy) would be a good idea.  i can shell this out, and let others make comments/suggestions/PR's.

arsenovic, 2017-01-04:
http://scikit-rf-web.readthedocs.io/about.html#

not a paper citation but here is a decent  blurb  about  references.

closed 2017-05-17 by arsenovic