scikit-rf/scikit-rf #75: Problem with network generation from impedance matrix

senselessDev, 2016-07-31:
```
import skrf
import numpy

lna = skrf.Network('test.s2p')

z = numpy.array([[100, -100], [-100, 100]])
a = z.reshape(1,2,2).repeat(201,0)

imp = skrf.Network(f = lna.f, z0 = lna.z0,  z = a, f_unit = 'Hz')
```

This is not working, because in my case `z` is set first. Then `z2s` is performed which wants to `fix_z0_shape` but it does not work. It seems the default value is set wrong:

```
Warning: Network has improper 'z0' shape.
Traceback (most recent call last):
  File "fail.py", line 9, in <module>
    imp = skrf.Network(f = lna.f, z0 = lna.z0,  z = a, f_unit = 'Hz')
  File "/usr/local/lib/python3.5/site-packages/skrf/network.py", line 387, in __init__
    self.__setattr__(attr,kwargs[attr])
  File "/usr/local/lib/python3.5/site-packages/skrf/network.py", line 1085, in z
    self._s = z2s(value, self.z0)
  File "/usr/local/lib/python3.5/site-packages/skrf/network.py", line 3564, in z2s
    z0 = fix_z0_shape(z0, nfreqs, nports)
  File "/usr/local/lib/python3.5/site-packages/skrf/network.py", line 4220, in fix_z0_shape
    elif len(z0)  == nports:
TypeError: len() of unsized object
```

arsenovic, 2016-08-01:
yes this is a bug. i think a `from_z` class method would make the most sense.  probably makes sense to construct from `s`, then assign `z` in this method.

arsenovic, 2016-08-01:
try that 6c7e9141b4b9e67c0391aba87c08b45a0d3e974d

senselessDev, 2016-08-03:
First of all, thanks for your great project and specially for your extreme fast response time! Really appreciating it :)

To topic:
Did not tried it yet (will see if I find some time to do so), but from my understanding the example above would not work. I don't know how the implementation of the whole framwork is done, but just changing `z =` in the example to 'from_z = ' won't work, right? I guess `skrf.Network.from_z()` would be the right way. 
Because I have used the framework just half a day by now, I'm not really used to it and its structure, but as a beginner I thought it would be nice if the shown example would work.

arsenovic, 2016-08-04:
Glad your enjoying skrf. Try the from_z constructor  and let me know if it works . 

Constructing from anything other than s parameters is less common , but we try not to impose unnecessary restrictions on the user .

arsenovic, 2016-08-11:
have you had a chance to try this?

senselessDev, 2016-08-16:
Yep. Working, Thanks!

```
import skrf
import numpy

lna = skrf.Network('test.s2p')

z = numpy.array([[100, -100], [-100, 100]])
a = z.reshape(1,2,2).repeat(201,0)

imp = skrf.Network.from_z(f = lna.f, z0 = lna.z0,  z = a, f_unit = 'Hz')
```

closed 2016-08-16 by senselessDev