scikit-rf/scikit-rf #78: se2gmm() documentation

jrmagers, 2016-10-03:
I have a 3-port single-ended network, let's call it a "differential power divider". Ideally, S21 and S31 should be -3 dB and angle(31)-angle(S21) = 180 degrees. 

I used se2gmm() to convert this network to mixed-mode parameters where port 1 remains single-ended and ports 2 and 3 become differential mode (100 ohms) and common mode (25 ohms).

se2gmm() seems to work fine (agrees with Microwave Office), but it does appear to be assuming that the ports to be transformed are 1 and 2. I was able to get around this by renumbering the ports.  This is  probably only an issue when there are an odd number of ports. If nothing else, it probably makes sense to document the behavior of se2gmm() because it wasn't completely clear.

```
import skrf as rf

ntwk = rf.Network("dpd.s3p")
ntwk.renumber([0,1,2],[2,1,0])
ntwk.se2gmm(p=1)
ntwk.renumber([2,1,0],[0,1,2])
```

arsenovic, 2016-10-03:
thanks for pointing this out, please feel free to submit a PR on this. 

you could also make an example notebook for this feature if you like??

jrmagers, 2016-12-15:
https://github.com/scikit-rf/scikit-rf/pull/82

closed 2016-12-15 by arsenovic