scikit-rf/scikit-rf #85: Plotting

hohe, 2017-01-03:
Since commit 29387cb9b83ce692d87045335125091c47c3078b "Moving Network Plotting Methods ..." I need to add 

skrf.plotting.setup_matplotlib_plotting()

after
import skrf

to my plotting python scripts. Is this the way I shall use the new plotting features? Is this documented somewhere?
I'm asking, because the examples are not working any more and I had to search some time for this solution.

arsenovic, 2017-01-03:
yes, this was a feature recently added, we will add docs to address this. (we also need to make the docs build again https://github.com/scikit-rf/scikit-rf/issues/81).  i think there is also a skrf.setup_pylab() command. @dvincentwest , can help .

hohe, 2017-01-03:
Ok, yes,
~~~~
skrf.setup_pylab()
~~~~
works also.
Thank you very much.

dvincentwest, 2017-01-03:
The changes were made with a vision to make skrf an available backend for doing the calculations while allowing for multiple data visualation platforms.  In addition to matplotlib, I would like to make bokeh available since it allows for user interaction.  Also this allows for the creation of some Qt-based widgets and gui applications that I have been making.

In addition, if you are not on the google group already (https://groups.google.com/forum/#!forum/scikit-rf) I will be sending out an announcement about the changes since documentation has been a little finicky.

hohe, 2017-01-04:
Ok, thank you.
I joined the google group. Your qt app looks really interesting.

closed 2017-01-10 by arsenovic