scikit-rf/scikit-rf #91: Changing networks frequency directly results in wrong data

vMeijin, 2017-01-10:
The documentation of Network.f specifies, that the returned frequency is in Hz. However if the unit of the Network is not Hz, changing the .f property results in wrong frequency data. 
Example: 
```
import skrf
import numpy

n = 5
sp = numpy.ones((n, 2, 2))
frequency = numpy.linspace(0, 1, n)
z0 = 50 * numpy.ones(n)

nw = skrf.Network(f_unit="GHz")
nw.s, nw.f, nw.z0 = sp, frequency, z0

print(nw.f)
nw.f = nw.f
print(nw.f)

>> [  0.00000000e+00   2.50000000e+08   5.00000000e+08   7.50000000e+08
   1.00000000e+09]
>> [  0.00000000e+00   2.50000000e+17   5.00000000e+17   7.50000000e+17
   1.00000000e+18]
```
If .f always returns the data in Hz it also should accept only data in Hz.

arsenovic, 2017-01-10:
yes this is correct. if you set/get `f` from the frequency property (ie `nw.frequency.f `) it works as you want 

the Network.f works differently which is misleading, and i think done for convenience. easy to change and probably makes sense to be consistent when reference the same property

arsenovic, 2017-01-10:
6aad82354b539f3557800eabaa59015e2059b15d

arsenovic, 2017-01-10:
so, it looks like a lot of tests rely on this feature through the network.__init__() function,  so i reverted it .  this will take more effort to fix all the tests.

Ttl, 2017-01-12:
Head seems to be still broken. f.setter doesn't even set anything so most of the things are failing.

arsenovic, 2017-01-12:
ok i look into this.

arsenovic, 2017-01-14:
@tt i forgot to revert, it should be working now.

alexpacini, 2019-03-22:
The commit is:
```python
self.frequency.f
```

why not
```python
self.frequency.f = f
```

I think the change was not setting the frequency at all.

arsenovic, 2019-04-06:
hi,  
  i chose this because i used ghz most when i originally wrote this. perhaps a poor choice, but i think its not worth changing due to backwards compatability. i  also regret making the Network.f property, because it just a pointer to  the Network.frequency.f property. there should be only one.

dkozel, 2019-08-30:
I think you should strongly consider putting these breaking changes on a roadmap for a future release. Simplicity of APIs and having "least surprise" defaults is worth folks having to add 'ghz' or 'frequency.' to a few lines of code after an update.

closed 2022-06-17 by jhillairet