scikit-rf/scikit-rf #92: Plotting in documentation examples is broken

Ttl, 2017-01-12:
Due to recent plotting changes it looks like that the all the examples in documentation that had plots are now failing.

Ttl, 2017-01-12:
I'm not too familiar with the plotting code, but would it be possible to define the plotting functions such that it would initialize the plotting with matplotlib when called the first time if no plotting library hasn't been initialized yet? It would avoid breaking the backwards compatibility while still allowing to use different plotting libraries.

@dvincentwest ?

arsenovic, 2017-01-12:
i am working on the docs now.

dvincentwest, 2017-01-12:
I can appreciate the nuisance of things changing, and also the nuisance of an extra line of code.  In addition to implementing the change to deal with conflicting versions of python-qt bindings, I also was hoping that to have a version of skrf that would have no plotting initialized at all.

I'm sue it is possible to automatically initiate the matplotlib plotting on condition of a couple of things:

- am I running in a notebook? (http://stackoverflow.com/questions/15411967/how-can-i-check-if-code-is-executed-in-the-ipython-notebook)

- use environment (or global) variables --> os.environ['skrf-autosetup-pylab'] = True/False, and decide what the default should be if no variable is set.  With this approach, the plotting could be initialized, but if I want to write a Qt-based program, I can turn it off before initializing skrf.

There certainly are ways to achieve what you suggest, I suppose its a question of design choice about what the default behavior should be?

arsenovic, 2017-01-14:
@ttl , i agree that having existing behavior work by default makes sense. 

@dvincentwest  has made this work in https://github.com/scikit-rf/scikit-rf/pull/95, which i think is a great solution. 

**on plotting**
eventually for interactive plotting, i think focusing on the notebook makes sense,   in which plotting can be handled like  graphical repr's

http://nbviewer.jupyter.org/github/scikit-rf/examples/blob/master/Network2.ipynb

here are lines which implement the notebook feature 

https://github.com/scikit-rf/scikit-rf/blob/master/skrf/network2.py#L290-L302

there is also a project called holoviews, which has a sort of self-visualization feature. i am hoping they integrate with xarray, and we can move the entire skrf backend into that. 

however, for a lot of apps  i think vinces qt approach  is great, and demonstrates that  separation of plotting from functional code is a the right way to go.

dvincentwest, 2017-01-14:
This looks good. I am going to propose a slight variation wherein rather
than a true false on pylab, the environment variable will specify which
plotting library to initialize where the default option is pylab. This will
allow others to be used if someone really wants it in the future and is
functionally the same as now. I will code it and send later today

http://nbviewer.jupyter.org/github/scikit-rf/examples/
blob/master/Network2.ipynb

https://github.com/scikit-rf/scikit-rf/blob/master/skrf/
network2.py#L290-L302

arsenovic, 2017-01-14:
( vince, you may want to modify your responses so they dont repeat the conversation )

arsenovic, 2017-01-14:
and yes, i think this makes sense. you are essentially creating a scikit-rf `backend`

dvincentwest, 2017-01-14:
Sorry about that, that's what happens when you reply directly from Gmail. I will check that in the future.

dvincentwest, 2017-01-14:
Pull request #94 has been updated with code changes to qtapps and the skrf/__init__.py implementing the os.environ['SKRF_PLOT_ENV'] environment variable, with 'pylab' being the default behavior if no variable is set.  Additionally, 'pylab-skrf-style' as an option will automatically initiate the skrf style by calling stylely()

closed 2017-01-19 by Ttl