scikit-rf/scikit-rf #102: Decide naming / conding conventions for scpi functions

dvincentwest, 2017-01-30:
SCPI is fairly straightforward ([SET commands and QUERY commands](https://en.wikipedia.org/wiki/Standard_Commands_for_Programmable_Instruments)) so the notion of having 1 proper method defined for each command is sort of self-defining.  Furthermore the convention for putting the command tree in a yaml file is also pretty simple:

Each Dict key represents either:

1. An SCPI command branch (CALC: {DATA/PAR/FORM, etc}
2. An SCPI command
3. both, in which case the dict key has the two explicity named objects: "command" and "branch".

In this way it is possible to write a "compiler" that parses the SCPI command tree in a recursive fashion and generates code that is always consistent in style and naming.

The main issues to decide:

1. What should be the names for all of the functions (currently they are rational PEP8 style names), but should they be based on something else, like some already established naming convention I am unaware of
2. Should the arguments be provided as args (arguments are always specified) or as kwargs (allowing us to decide default values for specific variables where appropriate?
3. What should the name of variables be: channel number = cnum, averaging_mode = mode, numerical argument = num, etc....
4. What should the naming convention of the scpi calls be.  Currently it is as follows

```
class SCPI():
    def set_pep8_scpi_name(kwarg1="", kwarg2=""):

    def query_pep8_scpi_name(kwarg1="", kwarg2=""):

class analyzer(SCPI):
```
so that a call would look something like:

```
analyzer.set_averaging_mode(cnum=1, mode="sweep")
analyzer.query_averaging_mode(cnum=1)
```

Is there a better way to be organizing this or it doesn't really matter and we should just get it going?
I'm going to leave this open for a few days before I start refining the idea in the code.

arsenovic, 2017-02-04:
i would always pick defaults, cnum=1 is the obvious one. dont a lot of the commands have defaults if you dont specify what they are?  this might be a choice like do you always specify a cnum, or if left None, does it just use 'current' ir active  channel?

 i would  might stay away from ambiguous things like `mode`,  even a `av_mode` for averaging, or `tgr_mode` for trigger or whatever. 

if you want i could start at the other end of this problem and help shell out the ABC for a generic analyzer. probably i will pull a lot from the current PNA class, but change some things. if you start this class where you want it, i can fill it in.

dvincentwest, 2017-02-04:
I have created a new branch in my repository.
https://github.com/dvincentwest/scikit-rf/tree/vna_drivers/skrf/vi

skrf.vi.vna is the logical place to put all this, but perhaps as a package rather than a module.  This obviously conflicts with the current vna module, but PNA and PNAX can be imported into the __init__.py module so the structure will be the same.

We can either rename the package or rename the old module to vna_legacy.py (or whatever).  I'll leave that to you to decide

The main VNA class in vna/vna.py was rapidly based on what I had started for the qtapps driver.  There are only a few methods, but it can be used as a context manager (which I like to avoid conflicts and unintentional lockout), and the __init__ method was based on my use case.  Also, it is only dependent upon pyvisa, and disregards python-ivi

Where it makes sense the method names should just be replaced with the names from the old vna module (haven't had time to go through that yet).  But if you structure the main vna class the way you think best, and based heavily on the old class, I will re-work the keysight_pna.PNA class to work with that VNA class.

Also take a look at the keysight PNA code.  The structure I started with was to make an SCPI object which contains all of the scpi functions so that they were all contained within a namespace.  Its easily changed if you think it is becoming too verbose, but conceptually I liked it.  The other way to do it is to just have the PNA class inherity the SCPI class so that it gains all of the functions.

As for keywork arguments, I need to rework the scpi_parser, but I think it should be fairly easy to implement something like:
CALC<cnum=1>, where the parser looks for a default value in the COMMAND_TREE and decides on a default value of "" if none is provided.

dvincentwest, 2017-02-04:
Recent push enables setting default values for scpi keyword arguments

cafeclimber, 2023-08-26:
I believe this can be closed with the latest update to the vi module

closed 2023-08-26 by jhillairet