scikit-rf/scikit-rf #103: py36 doesnt work

arsenovic, 2017-02-17:


RossWilliamson, 2017-03-09:
I cannot import skrf in python3.6:

import skfr as rf given:
`/home/rossw/anaconda/envs/py3k/lib/python3.6/site-packages/scikit_rf-0.14.3-py3.6.egg/skrf/network.py in read_touchstone(self, filename)
   1408         if self.name is None:
   1409             try:
-> 1410                 self.name = os.path.basename( os.path.splitext(filename)[0])
   1411                 # this may not work if filename is a file object
   1412             except(AttributeError):

/home/rossw/anaconda/envs/py3k/lib/python3.6/posixpath.py in splitext(p)
    118 
    119 def splitext(p):
--> 120     p = os.fspath(p)
    121     if isinstance(p, bytes):
    122         sep = b'/'

TypeError: expected str, bytes or os.PathLike object, not _io.TextIOWrapper
`

arsenovic, 2017-03-13:
this got fixed. py36 is now tested on travis.

closed 2017-03-13 by arsenovic