scikit-rf/scikit-rf #108: bug in determine_reflect code

dvincentwest, 2017-02-28:
In the skrf.calibration.determine_reflect function there is a bug on line 4017
```
    if reflect_approx is None:
        reflect_approx = reflect.copy()
        reflect_approx.s[:,0,0]=-1
```
there is no available reference to `reflect` in the function.  Currently all calling functions specify a value for `reflect_approx` so this code block is never entered.  The function should be re-written to either require an input value for reflect, or to simply require an input for reflect_approx instead of making it an optional None.  I don't rightly know what the right answer is

arsenovic, 2017-02-28:
yeah i think your interpretation is correct. the `reflect` is just used as a blank ntwk to copy info from, because immediately after copy() it's `s` parameters are set to -1. so it could be taken from any appropriate ntwk, like reflect_m, or thru_m. 

also the docstring are missing a `reflect_m`  entry.

arsenovic, 2017-02-28:
woops. yes your guess was right vince.

arsenovic, 2017-02-28:
you fix or me?

dvincentwest, 2017-02-28:
Now that I understand the perspective correctly and I will include in my upcoming commits

arsenovic, 2017-06-17:
hey vince, just closing old stuff out, is this fixed?

dvincentwest, 2017-06-17:
Yes

closed 2017-06-17 by arsenovic