scikit-rf/scikit-rf #111: .interpolate_from_f not working as intended

denzchoe, 2017-03-03:
Hi there,

A bit of a python newb here. Not sure how to use the lingo. So bear with me a bit.
I ran through examples from [here, scikit-rf.readthedocs](http://scikit-rf.readthedocs.io/en/latest/tutorials/Networks.html#References); and it appears that it is not working as intended.

Purpose, to match _line1_ to _line_ number of frequency points.
Over at the example the author used .resample for _line1_ to interpolate the frequency points of object _line1_ from 101 points to 201 points.
 
At the 2nd example, the author mentioned that .interpolate_from_f can be used too.
But it fails to work as intended.
Here is my example code

```
# To show example S-Parameter interpolation

import skrf as rf
from pylab import *
from matplotlib import pyplot as plt
#plots generated will have grids styled like RF plots
rf.stylely()

#import example data
from skrf.data import wr2p2_line1 as line1 
line = rf.data.wr2p2_line

print 'line:'
print line
print '\norig line1 :'
print line1 

#intepolate line1 here from 101 points to 201 points using .resample
line1.resample(201)
print '\nline1 via '  +"\x1b[32m\".resample\"\x1b[0m"
print line1

#intepolate line1 here from 101 points to 201 points using .resample
line1.resample(501)
print '\norig line1 again via '  +"\x1b[32m\".resample\"\x1b[0m"
print line1 

#intepolate line1 here from 101 points to 201 points using .interpolate_from_f
line1.interpolate_from_f(line.frequency)
print '\nline1 again via '  +"\x1b[31m\".interpolate_from_f\"\x1b[0m"
print line1 
```

and here's the output
```
line:
2-Port Network: 'wr2p2,line',  330-500 GHz, 201 pts, z0=[ 50.+0.j  50.+0.j]

orig line1 :
2-Port Network: 'wr2p2,line1',  330-500 GHz, 501 pts, z0=[ 50.+0.j  50.+0.j]

line1 via ".resample"
2-Port Network: 'wr2p2,line1',  330-500 GHz, 201 pts, z0=[ 50.+0.j  50.+0.j]

orig line1 again via ".resample"
2-Port Network: 'wr2p2,line1',  330-500 GHz, 501 pts, z0=[ 50.+0.j  50.+0.j]

line1 again via ".interpolate_from_f"
2-Port Network: 'wr2p2,line1',  330-500 GHz, 501 pts, z0=[ 50.+0.j  50.+0.j]
```

arsenovic, 2017-03-03:
hi denzchoe,
  the function `interpolate_from_f()` returns a ntwk, while `resample`
replaces the  network `in-place`.

so change
    line1.interpolate_from_f(line.frequency)
to
    line1 =  line1.interpolate_from_f(line.frequency)

also i highly recommend using jupyter notebook.  good luck!

alex

denzchoe, 2017-03-04:
Oh ok, I understand now...
Thanks for explaining!

closed 2017-03-04 by denzchoe

denzchoe, 2017-03-06:
By the way, a curious question.. how often will the scikit-rf package be updated on conda? I have noticed that you recently updated the multiline trl calibration code recently. Would be great to have them on Conda.

arsenovic, 2017-03-06:
we probably should shoot for a 6mo release schedule. but you can always download the latest from github, either through git or as a zip  (see button for 'clone or download'). once downloaded you can install within conda using the standard install method. 

1) activate your conda env 
2) cd the directory where you unzipped 
3) type  `python setup.py install`