scikit-rf/scikit-rf #112: Incorrect x-axis when plotting networks with different f_units

Ttl, 2017-03-05:
If two networks with different f_units are plotted in the same plot x-axis of the plot is incorrect.

A simple program with the bug:

    import skrf
    import matplotlib.pyplot as plt
    import numpy as np
    skrf.stylely()

    net1 = skrf.Network(s=np.random.random(100).reshape(-1,1,1), f=np.linspace(1e9,2e9,100), f_unit='Hz')
    net2 = skrf.Network(s=np.random.random(100).reshape(-1,1,1), f=np.linspace(1,2,100), f_unit='GHz')

    net1.plot_s_db()
    net2.plot_s_db()

    plt.show(block=True)

[image]

[image]

The networks have the same frequencies, but the first network with f_unit=Hz is plotted as if its f_unit was GHz.

arsenovic, 2017-03-05:
passing `f` to `Network()`,  is kind of a hack, passing a frequency object is preferred. regardless, this should not happen, as its unexpected. 

related : https://github.com/scikit-rf/scikit-rf/issues/91

arsenovic, 2017-03-06:
ok, i interpreted this error wrong. it looks like this is an error with the plot only, not the `f` attribute. 
`net1.frequency==net2.frequency` is `True`.

dvincentwest, 2017-09-15:
This appears to still be an open issue.  I hated dealing with this in some of the material characterization work I was doing so I ripped off the following approach from PyQtGraph.  In concept, you always pass the plotting data in the base units, and then auto-determine what the scaled si-unit should be.  In this case, the plotting function could be modified so that the frequencies are always plotted as Hz, and then you just change the tick-labels appropriately.  Here is the function I wrote for scaling the plot axes in my other project given below.

It just relies on [the SI scaling functions from the pyqtgraph functions.py module](https://github.com/pyqtgraph/pyqtgraph/blob/9d0779cc325b9a336674f20a7137a8c4fc09ada7/pyqtgraph/functions.py#L44)

Any thoughts?

```
def scale_frequency_axis(prefix=None, ax=None, funit=None):
    """
    scale the axis so that it is more readable (i.e. plot GHz instead of Hz)

    Parameters
    ----------
    prefix : str
        the SI prefix for Hz, k,M,G,T, etc
    ax : plt.axis
        the matplotlib axis object
    """
    if ax is None:
        ax = plt.gca()
    if ax.get_xscale() == "linear":
        if prefix is None:
            if funit is None:
                scale, prefix = si.siScale(ax.get_xlim()[1])
            else:
                if funit.lower() == "hz":
                    prefix = " "
                    scale = 1
                elif len(funit) == 3:
                    prefix = funit[0]
                    scale = si.SI_CONVERSION[prefix]
                else:
                    raise ValueError("invalid funit {}".format(funit))
        else:
            scale = si.SI_CONVERSION[prefix]
        ticks_x = ticker.FuncFormatter(lambda x, pos: '{0:g}'.format(x * scale))
        ax.xaxis.set_major_formatter(ticks_x)
        ax.set_xlabel("freq ({}Hz)".format(prefix))
    else:
        ax.set_xlabel("freq (Hz)")
```

Ttl, 2017-09-17:
Scaling the x-axis seems like a good idea, but I'm not very familiar with the plotting functions.

The same function could be used to change x-axis units to MHz/GHz when it makes sense. Currently plotting a network with f_unit Hz places a x10^9 mark on the right side when frequencies are over 1 GHz. At least I would have always preferred that units would be changed to GHz instead of keeping Hz and putting the x10^9 mark.

arsenovic, 2017-09-18:
vince, i think you are suggesting to always have plot data in Hz, so if wanted to pulled the data from the plot i always get units of Hz. but have ticks represent human readable units. i dont see a problem with this.

standing back for a second, this  looks to be an issue of not having an object for units.  if Network  was built on a  labeled nD-array, with  unit support, this would not be a  problem.

dvincentwest, 2017-09-18:
I have 2 commits in my repository:
https://github.com/dvincentwest/scikit-rf/commit/75c093ac8af7fe07f05af00d77105cdc7add8419

I can submit as a pull request, but I don't know how to test if it will screw anything up as far as proper plotting goes.  Any suggestions in this regard?

running @Ttl's example from above as this:
```
import skrf
import matplotlib.pyplot as plt
import numpy as np
skrf.stylely()

net1 = skrf.Network(s=np.random.random(100).reshape(-1,1,1), f=np.linspace(1e9,2e9,100), f_unit='Hz')
net2 = skrf.Network(s=np.random.random(100).reshape(-1,1,1), f=np.linspace(1,2,100), f_unit='GHz')

net1.plot_s_db(); net2.plot_s_db()

# 2nd plot with Hz
net2.plot_s_db(); net1.plot_s_db()
```
I get the plot in units of GHz as net2 was the last called network.  If you switch so that net1 is plotted last, it is plotted in units of Hz

[image]

[image]

Ttl, 2017-09-18:
I tried your changes and the frequency plots look good, but it breaks the units in time-domain plots. Time-domain plots are fixed to use ns, but after this change values are changed based on the funit while leaving the label as ns. The proper fix would be also handling the time units, but for a quick fix you could just add:

    if 'Time' not in xlabel:
        scale_frequency_ticks(ax, self.frequency.unit)

Ttl, 2017-09-19:
Fixed by @dvincentwest in pull request #179.

closed 2017-09-19 by Ttl