scikit-rf/scikit-rf #122: function convert_8term_2_12term only returns 10 terms

dvincentwest, 2017-04-13:
This came up as I was fiddling with an Agilent PNA calibration.  Performing a standard TRL calibration resulted in the following error terms being set on the VNA:

1. Directivity
2. Source Match
3. Reflection Tracking
4. Load match
5. Transmission tracking
6. Crosstalk

in the convert_8term_2_12term function, only terms 1-5 are return.
Also, internally skrf calls term 6 "isolation" instead of "crosstalk".

Can we just add in the following two lines at the end of convert_8term_2_12term:
```
    coefs_12term['forward isolation'] = npy.zeros_like(Elf)
    coefs_12term['reverse isolation'] = npy.zeros_like(Elf)
```
or would this cause any problems?

arsenovic, 2017-04-13:
so i think 8-term should have been called 10-term, and isolation should have been include-able. 
@Ttl, do you agree? are you using isolation (aka crosstalk)  in your TRL?

we might consider adding isolation terms into the `coefs_list_8term`. but, for the  conversion function you need, you should test if there is no isolation term present, and if not, fill  it with zeros like you suggest

Ttl, 2017-04-13:
None of the EightTerm calibrations use the isolation terms. EightTerm model can be extended to include them, but using them would require some changes in the code. For example switch correction is calculated in a different way if the isolation terms aren't zero.

I don't think that including the zero isolation terms in the convert_8term_2_12term as you suggested will break anything. In fact I think it's a bug that they are not initialized to zero because 12-Term calibrations do initialize them to zero if isolation calibration is not done.

arsenovic, 2017-04-14:
ok, sounds  like maybe we should make a new issue  for adding isolation to the 8term class of calibrations, and close this one. 
#123

closed 2017-04-14 by arsenovic