scikit-rf/scikit-rf #123: Add isolation (crosstalk) to EightTerm Class of calibrations,

arsenovic, 2017-04-14:
The class of EightTerm calibrations should allow for  isolation correction terms. 
Obviously, this would make the class more appropriately named `TenTerm`!, but  i think leaving it as EightTerm is ok, even though it's misleading. 

see #122,

dvincentwest, 2017-04-14:
How do you want to do this?  There are 4 classes of EightTerm Coefficients, and they each define their own self.run method where the coefs dictionaries are defined.

Two proposals:
1) put it in EightTerm.coefs getter
    If no isolation terms are detected, initialize at zero and return
2) put it in the EightTerm..coefs setter
    If not isolation terms are detected, initalize at zero and set the _coefs attribute with this

Or something else?

arsenovic, 2017-04-15:
so  i think we want to 

 1) change the init to look for isolation temrs as a kwarg to `EightTerm`, so that we could use it in the correction. 
 2) use it in the correction (`EightTerm.apply_cal`). this may require terminate/unterminate  to be updated as @Ttl  suggests. 
  3) use it in `EightTerm.embed`. (low priority, but needed for tests)
  3) have default zero-filling fall-back behavior as you suggest above.

arsenovic, 2017-04-15:
also putting the isolation in coefs getter, as you suggest solves the problem is redundant implementation, which  is important . after looking the `EightTerm.run` directly accesses self._coefs, so it bypasses the setter. so going with the getter makes the most sense.

dvincentwest, 2017-04-18:
are we ready to close this then?

arsenovic, 2017-04-18:
Yep

closed 2017-04-18 by dvincentwest