scikit-rf/scikit-rf #126: RectangularWaveguide impedance calculation is wrong

PaulKGrimes, 2017-04-19:
The impedance calculation for RectangularWaveguide has an error that results in propagating modes returning imaginary impedances from the Z0() method, and the mismatch between a TE mode and free space giving an insertion loss greater than 1.

Pozar eq. (3.86) gives the impedance of a TE mode as Z_TE = k * eta / beta  where beta is real for propagating modes and imaginary for evanescent modes.  The source code shows that the RectangularWaveguide impedance is calculated using gamma, which is defined as being imaginary for propagating modes, which in turn gives imaginary impedances for propagating modes.

arsenovic, 2017-04-19:
you are correct this is an embarrassing bug. fixed in  e267428. let me know if this gives you the results you expect. the forumla in skrf is from Roger Harrington - Time Harmonic Electromagnetic Fields, section 8.1

PaulKGrimes, 2017-04-19:
Thanks, when I manage to fix my Anaconda install I'll check the results.  My only concern is that just multiplying by 1j might give the wrong sign for attenuation, but that's easily checked.

arsenovic, 2017-04-19:
you can also download a zip of the master branch from github, if thats easier

arsenovic, 2017-04-24:
paul, after some testing, the current expression appeared to give incorrect results for attenuation, in terms of magnitude of Z0. so,  i reverted the expression back to the original one, and this appears to  give correct results.  consistently choosing where to put the 'j' causes a lot of bugs!

we will make a release soon and you test it,

arsenovic, 2017-05-17:
@PaulKGrimes, the newest release of skrf is out with this fix you should be able to install with 

`conda install scikit-rf -c conda-forge`

let me know if my change is  correct.

arsenovic, 2017-06-17:
@PaulKGrimes , can you test to see if this is fixed, and i will close this issue.

PaulKGrimes, 2017-06-19:
Hi Alex,

That does seem to have fixed the issue!  Thanks.

closed 2017-06-19 by PaulKGrimes