scikit-rf/scikit-rf #128: skrf.Network fail to read version 2.0 touchstone file

ring630, 2017-04-25:
Hi,
as stated, skrf.Network fail to read version 2.0 touchstone file. 

the s4p file to be read

```
! Touchstone data file
[Version] 2.0
# Hz S MA R 50
[Number of Ports] 4
[Number of Frequencies] 3501

[Network Data]
! freq                S11mag               S11ang                S12mag               S12ang                S13mag               S13ang                S14mag               S14ang              
!                     S21mag               S21ang                S22mag               S22ang                S23mag               S23ang                S24mag               S24ang
```
the error I got
```
C:\Anaconda3\lib\site-packages\skrf\io\touchstone.py in load_file(self, fid)
    180             # collect all values without taking care of there meaning
    181             # we're seperating them later
--> 182             values.extend([ float(v) for v in line.split() ])
    183 
    184         # let's do some post-processing to the read values

C:\Anaconda3\lib\site-packages\skrf\io\touchstone.py in <listcomp>(.0)
    180             # collect all values without taking care of there meaning
    181             # we're seperating them later
--> 182             values.extend([ float(v) for v in line.split() ])
    183 
    184         # let's do some post-processing to the read values

ValueError: could not convert string to float: '[number'
```

arsenovic, 2017-04-25:
can you provide me and an example of such a file?

JimInCO, 2017-04-25:
It appears to me that the Touchstone reader is not fully implementing some of the comments that can be used in [version 2.0](https://ibis.org/touchstone_ver2.0/touchstone_ver2_0.pdf).
Page 5 of the version 2.0 document shows several required and optional keywords. It appears to me that the code is only handling [version] and [reference]

ring630, 2017-04-25:
@JimInCO I have the same view. the reader looks for data directly after _# Hz S MA R 50_. it is correct if the file is saved as touchstone version 1.x .
Version 2.0 defined optional keywords as below, but the reader treats them as data.
```
[Number of Ports] 4
[Number of Frequencies] 3501

[Network Data]
```

arsenovic, 2017-04-25:
discarding this information is easy. reading them would take a little bit of  work. would either of you be interested  to work on this?

dvincentwest, 2017-04-26:
Also worth mentioning, Touchstone v 2.0 allows to treat each port independently with regard to reference environment, it will definitely take a lot attention and work to be compatible with that.  Reading in the variables is probably a simple regex, but writing the logic to apply those variables to each port independently (if necessary) is required to be fully compatible with Touchstone v. 2.0

arsenovic, 2017-04-26:
not relevant to reading this file format, but worth keeping in mind that  a  complete description of a network's ports requires every frequency to define a complex  characteristic impedance and propagation constant.

arsenovic, 2017-05-12:
so given the underwelling response to implementation, would anyone be interested to fund this feature? otherwise, i will leave the issue open for now.

JimInCO, 2017-05-13:
If someone can get me a sample file I can work on it.

Thanks,
James

arsenovic, 2017-05-16:
@ring630 , do you have a sample file for @JimInCO ?

ring630, 2017-05-16:
@JimInCO ,here is an example from Touchstone 2.0 [Spec.](https://ibis.org/touchstone_ver2.0/touchstone_ver2_0.pdf)
[example.zip](https://github.com/scikit-rf/scikit-rf/files/1005057/example.zip)

Thanks for your help!

arsenovic, 2017-06-17:
any progress on this?

jhillairet, 2017-08-29:
The issue should be resolved with PR#170. To be tested with various .ts files.

closed 2017-08-29 by arsenovic