scikit-rf/scikit-rf #133: Can't install with Python3.6 on Anaconda

kemiktau, 2017-05-16:
Here is the error I get:
UnsatisfiableError:
The following specifications were found to be in conflict:
- python ==3.6.1
- scikit-rf -> python 2.7*
Use "conda info " to see the dependencies for each package.

arsenovic, 2017-05-16:
what channel are you installing from? anaconda, conda-forge, or scikit-rf?

kemiktau, 2017-05-16:
Anaconda.

I had Python 3.6 and when I tried to install scikit-rf on anaconda, I got he above error. I downgraded Python to 3.5 and tried again and installation worked. Now, when I try to upgrade to my Python to 3.6, I, obviously, get the same error.

arsenovic, 2017-05-16:
try to install from scikit-rf channel 

conda install scikit-rf -c scikit-rf

kemiktau, 2017-05-17:
Works. Thank you

closed 2017-05-17 by arsenovic