scikit-rf/scikit-rf #138: N-ports HFSS Touchstone file

jhillairet, 2017-07-11:
Dear all,

It's a long time since I've used skrf, so I wonder if the following is a problem or a misuse. 

I've got a Touchstone file of an 8-ports device(*) which has been exported from HFSS in the Magnitude/Angle format. One file has been exported with the 50Ohm renormalization while not the other (remove the .txt extension): 
[HFSS_S_MA_example_50Ohm.s8p.txt](https://github.com/scikit-rf/scikit-rf/files/1138365/HFSS_S_MA_example_50Ohm.s8p.txt)
[HFSS_S_MA_example.s8p.txt](https://github.com/scikit-rf/scikit-rf/files/1138367/HFSS_S_MA_example.s8p.txt)

It is a detail, but the network frequency display is rounded. Is it intended?
```
nwk = rf.Network('HFSS_S_MA_example.s8p')
nwk_50Ohm = rf.Network('HFSS_S_MA_example_50Ohm.s8p')
nwk.frequency.f
Out[125]: array([  3.70000000e+09])
nwk.frequency
Out[124]: 3-3 GHz, 1 pts
```

In HFSS, when exporting the Touchstone file, there is the possibility (now by default) to add the gamma and port impedances (for each ports) in the comments, which I think is great since it is always a source of confusion when analyzing some files. Presently these informations are not parsed (since I guess such additional info are not standard in touchstone files):
```
nwk.z0
Out[177]: 
array([[ 50.+0.j,  50.+0.j,  50.+0.j,  50.+0.j,  50.+0.j,  50.+0.j, 50.+0.j,  50.+0.j]])

nwk_50Ohm.z0
Out[178]: 
array([[ 50.+0.j,  50.+0.j,  50.+0.j,  50.+0.j,  50.+0.j,  50.+0.j, 50.+0.j,  50.+0.j]])
```
while for the former case, one would expect `z0=445.259695889026+1.51087559077766j `, as it is written in the Touchstone file. As a direct consequence, skrf renormalize the S parameters to 50hm, which is not the correct value here, so the S param are not as expected. 

_Tricky_: in the HFSS touchstone file, the port impedances are written in real/imaginary, whatever the format used for the scattering parameters (MA, DB, RI)

Which brings me to the following : since I guess the way HFSS is formatting the Touchstone file is certainly not standard, the `rf.hfss_touchstone_2_network()` is definetly of interest. However presently it does not work with such files (returns a `ValueError: could not convert string to float: '!' `). 

Before trying to patch the code myself, I would like to discuss if the present behavior is normal/expected. 

(*) : all 8 ports are rectangular waveguides and thus setup as "Z_wave" ports in the HFSS terminology.

arsenovic, 2017-07-12:
hi julien, 
      about frequency rounding. if you are referring to the repr which is printed after ntwk, then yes it should be  rounded, just for concision.  this could be done more intelligently though.

about z0. skrf will automatically set z0 to 50, unless you specify what it is. this is not the same as z0-re-normalization, but you can do that as well.  if you set you ntwk.z0 to 445, and then use ntwk.[renormalize(50)](http://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.renormalize.html#skrf.network.Network.renormalize),  then ntwk and ntwk_50ohm should match.  

fixing  `hfss_touchstone_2_network() ` would be much appreciated. it broke a while ago, because they changed their format, so you may want to do some version detection. 

i think lack of standardized  z0/gamma support is a major flaw in the touchstone format.

jhillairet, 2017-07-17:
Hi, 

New question : should `Network()` detects HFSS touchstone files and parse it with its dedicate function?

The difference is that touchstone reading function used under the hood by `Network()` does not parse the impedance and gamma, while `hfss_touchstone_2_network()`

arsenovic, 2017-07-17:
i think this would be a great feature.

jhillairet, 2019-05-10:
Closed with PR #148

closed 2019-05-10 by jhillairet