scikit-rf/scikit-rf #139: 'list' object has no attribute 'real' in calibration.NISTMultilineTRL

skuep, 2017-07-13:
Hi,
I am trying the NIST Multiline TRL and I am stuck with the error given in the title. The corresponding traceback says:
```
Traceback (most recent call last):
  File "Losses.py", line 34, in <module>
    calInst.run()
  File "C:\ProgramData\Anaconda3\lib\site-packages\skrf\calibration\calibration.py", line 2728, in run
    e_val = root_choice(Mij, dl, gamma_est)
  File "C:\ProgramData\Anaconda3\lib\site-packages\skrf\calibration\calibration.py", line 2686, in root_choice
    and ga.real > 0:
AttributeError: 'list' object has no attribute 'real'
```

The corresponding line in calibration.py seems to me obvious why this fails
```
                if abs((ga[0]-gb[0]).real) < 0.1*abs((ga[1] + gb[1]).real) \
                    and abs(ga[0].real/ga[0].imag) > 0.001 \
                    and ga.real > 0:
```
There is an object "ga", which is a list (thus the indirections in the first line). But on the third line, the attribute ga.real gets called, which does not exist in a list item.

dvincentwest, 2017-07-13:
I haven't looked closely yet, but my assumption is that ga should not be a list but either a complex number or a numpy array of complex numbers.  Can you provide the code snippet you have that calls the calibration?  There may be an error

dvincentwest, 2017-07-13:
I retract my statement, and I believe you are correct.  ga and gb are length-2 lists of complex numbers, and probably the line ga.real should be ga[0].real.  @Ttl, can you confirm that is the correction or are we missing something?

Ttl, 2017-07-13:
Yes, it should be ga[0].real. I'll submit a pull request.

I haven't hit this line of the code ever. You might want to double check the data and calibration routine arguments since it looks like the calibration routine has serious difficulties in determining the correct root.

skuep, 2017-07-19:
Thanks everyone.
I am no math-buff (just an engineer ;-)), but I already encountered the problem you are referring to. The algorithm does not work very well, because it cannot find the roots (as is displayed in the terminal).

dvincentwest, 2017-09-14:
@arsenovic , @Ttl I believe we can close this issue no?

arsenovic, 2017-09-15:
ill defer  to henrik on this

Ttl, 2017-09-15:
This was fixed in the pull request #140.

closed 2017-09-15 by arsenovic