scikit-rf/scikit-rf #144: Data file for examples location

razvanc1, 2017-08-01:
New in scikit-rf and github and I am trying to run 

import skrf as rf
from skrf.data import ring_slot
ring_slot = rf.Network('data/ring slot.s2p')

I got error bellow . Can you help me?

runfile('C:/Razvan\'s/python RF/scikit_rf tutorials/test1.py', wdir='C:/Razvan\'s/python RF/scikit_rf tutorials')
Traceback (most recent call last):

  File "<ipython-input-65-b1fde78e5eaf>", line 1, in <module>
    runfile('C:/Razvan\'s/python RF/scikit_rf tutorials/test1.py', wdir='C:/Razvan\'s/python RF/scikit_rf tutorials')

  File "C:\ProgramData\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 880, in runfile
    execfile(filename, namespace)

  File "C:\ProgramData\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Razvan's/python RF/scikit_rf tutorials/test1.py", line 11, in <module>
    ring_slot = rf.Network('data/ring slot.s2p')

  File "C:\ProgramData\Anaconda3\lib\site-packages\skrf\network.py", line 379, in __init__
    fid = get_fid(file, 'rb')

  File "C:\ProgramData\Anaconda3\lib\site-packages\skrf\util.py", line 166, in get_fid
    return open(file, *args, **kwargs)

FileNotFoundError: [Errno 2] No such file or directory: 'data/ring slot.s2p'

arsenovic, 2017-08-01:
hi, 
try this instead, 
`from skrf.data import ring_slot` 

 and work from there.   that example depends on you being in the correct working directory, which is somewhere in the scikit-rf package.  if eventually you have touchstone data you will know where it is.

arsenovic, 2017-08-01:
woops,  i didnt read your whole code. looks like you do import ring_slot and then trying to load it from disk.  just remove the rf.Network(...) line

razvanc1, 2017-08-02:
thank you.
Based on your indication I was able to load a 16p file . Please see bellow
import skrf as rf
from skrf.data import ring_slot
#ring_slot = rf.Network('data/ring slot.s2p')

import os
#touchstone_file = os.path.join(rf.data.pwd,’ring slot.s2p’)

touchstone_file = '/test/test.s16p'
ring_slot = rf.Network(touchstone_file)

ring_slot
Out[77]: 16-Port Network: 'test',  10000000-20000000000 Hz, 1601 pts, z0=[ 50.+0.j  50.+0.j  50.+0.j ...,  50.+0.j  50.+0.j  50.+0.j]

arsenovic, 2017-08-02:
excellent.

closed 2017-08-02 by arsenovic

razvanc1, 2017-08-07:
I like much to use scikit-rf for deembeding a multiport ( 10 – 20 ports) device.
I have , let see s11p file and corresponding 11  s2p files, for each trace. 
Based on manual I tested successfully , deembeding for a 2 port device but I am wondering if you can help me with my problem.
I attached here my script for 2 port device
import skrf as rf
#import matplotlib.pyplot as plt
#%matplotlib inline
from pylab import *
rf.stylely()
from skrf.data import ring_slot
#ring_slot = rf.Network('data/ring slot.s2p')

import os
#touchstone_file = os.path.join(rf.data.pwd,’ring slot.s2p’)

touchstone_file = '/test/Bandpass Filter.s2p'
touchstone_file2 = '/test/3 dB Pad R.s2p'
ring_slot = rf.Network(touchstone_file)
ring_slot2 = rf.Network(touchstone_file2)
ring_slot.plot_s_db(m=0,n=1)
ring_slot2.plot_s_db(m=0,n=1)
#ring_slot.plot_s_smith()
t1= rf.Network.a_db

figure()
ring_slot.plot_s_db(m=0,n=0)
ring_slot2.plot_s_db(m=0,n=0)

figure()
ring_slot.plot_s_db(m=1,n=0)
ring_slot2.plot_s_db(m=1,n=0)

deemb1 = ring_slot.inv ** ring_slot2
deemb2 = ring_slot2.inv ** ring_slot

figure()
ring_slot.plot_s_db(m=1,n=0, label = 'S21 measured')
#ring_slot2.plot_s_db(m=1,n=0)
deemb1.plot_s_db(m=1,n=0, label = 'S21 deembeded')
#deemb2.plot_s_db(m=1,n=0)

figure()
ring_slot.plot_s_db(m=1,n=1, label = 'S22 measured')
#ring_slot2.plot_s_db(m=1,n=0)
deemb1.plot_s_db(m=1,n=1, label = 'S22 deembeded')
#deemb2.plot_s_db(m=1,n=0)