scikit-rf/scikit-rf #180: touchstones load error in py3.6 on windows

arsenovic, 2017-09-25:
from oliver david mailing list post:

Hello,
I was coding in Python 2.7 (with Python xy tools) and I added Sciki RF package to do deembedding calculation (loss and phase from our measurement). I was working perfectly.

But now, we migrate to Python 3.6 (WinPython tools). In a simple file I just do an import, and it crash as:

Traceback (most recent call last):
  File "C:\WinPython\WinPython-32bit-3.6.2.0Qt5\python-3.6.2\lib\site-packages\skrf\network.py", line 368, in __init__
    self.read(fid)
  File "C:\WinPython\WinPython-32bit-3.6.2.0Qt5\python-3.6.2\lib\site-packages\skrf\network.py", line 1870, in read
    self.copy_from(read(*args, **kwargs))
  File "C:\WinPython\WinPython-32bit-3.6.2.0Qt5\python-3.6.2\lib\site-packages\skrf\io\general.py", line 104, in read
    obj = pickle.load(fid, *args, **kwargs)
_pickle.UnpicklingError: invalid load key, '!'.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:/LaboPython/Projects3x/_Empty/Script/TOS.py", line 1, in <module>
    import skrf as rf
  File "C:\WinPython\WinPython-32bit-3.6.2.0Qt5\python-3.6.2\lib\site-packages\skrf\__init__.py", line 24, in <module>
    from . import media
  File "C:\WinPython\WinPython-32bit-3.6.2.0Qt5\python-3.6.2\lib\site-packages\skrf\media\__init__.py", line 60, in <module>
    from .rectangularWaveguide import RectangularWaveguide
  File "C:\WinPython\WinPython-32bit-3.6.2.0Qt5\python-3.6.2\lib\site-packages\skrf\media\rectangularWaveguide.py", line 13, in <module>
    from ..data import materials
  File "C:\WinPython\WinPython-32bit-3.6.2.0Qt5\python-3.6.2\lib\site-packages\skrf\data\__init__.py", line 26, in <module>
    ntwk1 = Network(os.path.join(pwd, 'ntwk1.s2p'))
  File "C:\WinPython\WinPython-32bit-3.6.2.0Qt5\python-3.6.2\lib\site-packages\skrf\network.py", line 374, in __init__
    self.read_touchstone(fid)
  File "C:\WinPython\WinPython-32bit-3.6.2.0Qt5\python-3.6.2\lib\site-packages\skrf\network.py", line 1617, in read_touchstone
    self.name = os.path.basename( os.path.splitext(filename)[0])
  File "C:\WinPython\WinPython-32bit-3.6.2.0Qt5\python-3.6.2\lib\ntpath.py", line 223, in splitext
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not _io.TextIOWrapper

After investigate, I found the issue in the loading file s2p for the example. I removed all import from data and cleaned the file __init__.py from this reference, and the import work well.

In a second step, I wanted to load my own S2P file, but it cause exactly the same error. I understand the issue is when loading the Header of th S2P, but I don't understant how it can be write on web site it test on Python 3.6, it's not working.

arsenovic, 2017-09-25:
he had old version

closed 2017-09-25 by arsenovic

leochand101, 2022-08-25:
Hey I have a similar error with the following output, any pointers on conversion from old to newer format would be welcome.
`!*************************************************************************
!! ASITIC Ver Grackle
!!  Data generated by the <2Port> command.
!! Calculation mode, fast= TRUE
!! Generated on Wed Jul  2 11:57:28 1975
!# Spiral Data:	Name = <a>, Square Spiral Inductor
!# L1 = 175.00, L2 = 175.00, Origin = <200.00,200.00>
!# Sides =   4, Metal =   6, Exit Metal =   5
!# W = 10.00, S =  3.50, N =  5.00
!# Theta =  0.00, Phase =   1
!# Technolog File Data <sky130.tek> @ <.>
!# Auto-Cell Mode:  <alpha> = 0.050000, <beta> = 0.800000
!# Metal Layers
!# Name=MSUB      Met=0  Lyr=  1  Mz=11.98  T=0.10  RS=0.03  Col=RED     
!# Name=LI        Met=1  Lyr=  3  Mz=10.54  T=0.10  RS=12.80  Col=BLUE    
!# Name=MET1      Met=2  Lyr=  5  Mz=10.13  T=0.36  RS=0.12  Col=GREEN   
!# Name=MET2      Met=3  Lyr=  6  Mz=9.50  T=0.36  RS=0.12  Col=YELLOW  
!# Name=MET3      Met=4  Lyr=  7  Mz=8.72  T=0.84  RS=0.05  Col=ORANGE  
!# Name=MET4      Met=5  Lyr=  8  Mz=7.48  T=0.84  RS=0.05  Col=GREY    
!# Name=MET5      Met=6  Lyr=  9  Mz=6.13  T=1.26  RS=0.03  Col=WHITE   
!# Substrate Layers
!# Lyr=    0  Eps=  12  Rho=1e+03  DZ=  22  T=  10
!# Lyr=    1  Eps=  12  Rho=1.5e+05  DZ=  12  T=   1
!# Lyr=    2  Eps= 3.9  Rho=1e+14  DZ=  11  T=0.94
!# Lyr=    3  Eps= 7.3  Rho=1e+14  DZ=  11  T=0.15
!# Lyr=    4  Eps=   4  Rho=1e+14  DZ=  10  T=0.27
!# Lyr=    5  Eps= 4.5  Rho=1e+14  DZ=  10  T=0.63
!# Lyr=    6  Eps= 4.2  Rho=1e+14  DZ= 9.5  T=0.78
!# Lyr=    7  Eps= 4.1  Rho=1e+14  DZ= 8.7  T= 1.2
!# Lyr=    8  Eps=   4  Rho=1e+14  DZ= 7.5  T= 1.4
!# Lyr=    9  Eps= 7.5  Rho=1e+14  DZ= 6.1  T= 1.3
!# Lyr=   10  Eps= 3.9  Rho=1e+14  DZ= 4.9  T=0.09
!# Lyr=   11  Eps= 7.5  Rho=1e+14  DZ= 4.8  T=0.54
!# Lyr=   12  Eps= 2.9  Rho=1e+14  DZ= 4.2  T= 4.2
!! Ind Cell size:  L = 256.00 	 W = 256.00 	 T = 10.00
!! Cap Cell size:  L = 256.00 	 W = 256.00
!*************************************************************************
# HZ S MA RS11 mag    S11 phase  S12 mag    S12 phase  S21 mag    S21 phase  S22 mag    S22 phase  
9.999999869911040e+14  1.152085364155556e+00  1.798625589632711e+02  1.280149848057670e-01  -6.687495856013712e-01  1.280149848057670e-01  -6.687495856013712e-01  8.719953287141892e-01  -1.798975986355379e+02
1.999999973982208e+15  1.152021346585857e+00  1.799312480186079e+02  1.280037467587700e-01  -3.344119478626505e-01  1.280037467587700e-01  -3.344119478626505e-01  8.719988319714808e-01  -1.799487976701811e+02
3.000000028082176e+15  1.152009487802567e+00  1.799541614600518e+02  1.280016652679945e-01  -2.229458869766647e-01  1.280016652679945e-01  -2.229458869766647e-01  8.719994808631761e-01  -1.799658649099976e+02
3.999999947964416e+15  1.152005336964453e+00  1.799656200749548e+02  1.280009367200249e-01  -1.672106198273306e-01  1.280009367200249e-01  -1.672106198273306e-01  8.719997079844368e-01  -1.799743986290750e+02
5.000000136282112e+15  1.152003415675335e+00  1.799724956822623e+02  1.280005995021531e-01  -1.337689418747512e-01  1.280005995021531e-01  -1.337689418747512e-01  8.719998131103848e-01  -1.799795188834702e+02
6.000000056164352e+15  1.152002372001322e+00  1.799770795642736e+02  1.280004163210902e-01  -1.114743200529156e-01  1.280004163210902e-01  -1.114743200529156e-01  8.719998702160277e-01  -1.799829323939768e+02
6.999999976046592e+15  1.152001742695547e+00  1.799803538237505e+02  1.280003058684156e-01  -9.554952166853750e-02  1.280003058684156e-01  -9.554952166853750e-02  8.719999046489322e-01  -1.799853706187629e+02
7.999999895928832e+15  1.152001334250804e+00  1.799828095457195e+02  1.280002341803069e-01  -8.360589068953916e-02  1.280002341803069e-01  -8.360589068953916e-02  8.719999269971783e-01  -1.799871992887427e+02
8.999999815811072e+15  1.152001054222050e+00  1.799847195655504e+02  1.280001850311603e-01  -7.431638337627498e-02  1.280001850311603e-01  -7.431638337627498e-02  8.719999423190263e-01  -1.799886215884276e+02
1.000000027256422e+16  1.152000853919017e+00  1.799862475893190e+02  1.280001498750685e-01  -6.688476832301843e-02  1.280001498750685e-01  -6.688476832301843e-02  8.719999532786267e-01  -1.799897594285503e+02
!*************************************************************************
!! ASITIC Ver Grackle
!!  Data generated by the <2Port> command.
!! Calculation mode, fast= TRUE
!! Generated on Tue Sep  3 03:12:40 1974
!# Spiral Data:	Name = <a>, Square Spiral Inductor
!# L1 = 175.00, L2 = 175.00, Origin = <200.00,200.00>
!# Sides =   4, Metal =   6, Exit Metal =   5
!# W = 10.00, S =  3.50, N =  5.00
!# Theta =  0.00, Phase =   1
!# Technolog File Data <sky130.tek> @ <.>
!# Auto-Cell Mode:  <alpha> = 0.050000, <beta> = 0.800000
!# Metal Layers
!# Name=MSUB      Met=0  Lyr=  1  Mz=11.98  T=0.10  RS=0.03  Col=RED     
!# Name=LI        Met=1  Lyr=  3  Mz=10.54  T=0.10  RS=12.80  Col=BLUE    
!# Name=MET1      Met=2  Lyr=  5  Mz=10.13  T=0.36  RS=0.12  Col=GREEN   
!# Name=MET2      Met=3  Lyr=  6  Mz=9.50  T=0.36  RS=0.12  Col=YELLOW  
!# Name=MET3      Met=4  Lyr=  7  Mz=8.72  T=0.84  RS=0.05  Col=ORANGE  
!# Name=MET4      Met=5  Lyr=  8  Mz=7.48  T=0.84  RS=0.05  Col=GREY    
!# Name=MET5      Met=6  Lyr=  9  Mz=6.13  T=1.26  RS=0.03  Col=WHITE   
!# Substrate Layers
[... 50 lines cut ...]
!# Lyr=    2  Eps= 3.9  Rho=1e+14  DZ=  11  T=0.94
!# Lyr=    3  Eps= 7.3  Rho=1e+14  DZ=  11  T=0.15
!# Lyr=    4  Eps=   4  Rho=1e+14  DZ=  10  T=0.27
[... cut ...]
7.999999895928832e+15  -1.152001282400366e+00  3.456349863576765e-04  1.280000979075059e-01  -1.867776279565949e-04  1.280000979075059e-01  -1.867776279565949e-04  -8.719999052346480e-01  -1.948174766777347e-04
8.999999815811072e+15  -1.152001013253649e+00  3.072316414209544e-04  1.280000773587565e-01  -1.660245874403918e-04  1.280000773587565e-01  -1.660245874403918e-04  -8.719999251239356e-01  -1.731711175577788e-04
1.000000027256422e+16  -1.152000820734522e+00  2.765088254394447e-04  1.280000626603823e-01  -1.494221476349061e-04  1.280000626603823e-01  -1.494221476349061e-04  -8.719999393506002e-01  -1.558540237002606e-04`

jhillairet, 2022-08-27:
> Hey I have a similar error with the following output, any pointers on conversion from old to newer format would be welcome.

This seems normal, as the file format is not a single Touchstone file but rather a collection of Touchstone-like files. I guess it should work if you split the data into separate files.