scikit-rf/scikit-rf #183: SCPI parser should have limited universal name parameters implemented

dvincentwest, 2017-10-31:
@arsenovic 
@helgrind 
@beuerle 

Maybe this never goes very far, but the SCPI parser has the guts of something really nice.  Some parts of the process could be more automated and enforced I think.  There are still only a few instruments added, but many thanks to @beuerle  for implementing the ZVA scpi drivers.  I have been augmenting the main driver with some meta functions including get_snp_network which will also enable get_oneport and get_twoport (PR soon).  It seems that a good idea would be to have a template scpi yaml file with default function name parameters:

```yaml
CALC:
  DATA: data
  PAR: {
    SEL: selected_measurement
  } ... etc
```

The parser would then read this first, and then when it encounters such a branch in the _scpi.yaml files it would do one of several possible things:

1) ignore any name parameter set for these functions
1) warn that name was assigned to a reserved branch
1) allow overwriting the name, but with the warning (I think I am against this)
1) warn that a name was in-conflict and re-write the yaml file to match (this sounds scary, maybe hard to do, but maybe the most productive)
1) raise an exception, but scan the whole tree for name conflicts and print a list and demand the user change all the names to match the validation file before compiling the scpi.py file

The second this requires is some kind of convention.  The is related to issue #102 but would be a good solution for that I think.  Possibly I will set to making a yaml file, but would very much appreciate some discussion around what names for different parameters should be.  I have very little experience with SCPI instruments outside of Network Analyzers (none that I can think of) and I may not be the best person to recognize when CALC:SEL:PAR corresponds to something wildly different in different instruments for example.  This kind of name enforcement really should only apply to SCPI commands that perform nearly universal functions from instrument to instrument.

Thoughts?

arsenovic, 2017-10-31:
in my experience most skrf users care most about VNA's. and i would venture a guess to say that 90% of VNA users just want clear and easy methods to get the network data (snp), and 90% users probably have R&S or Keysight/Agilent/HP.

i would always allow for custom stuff, so i am in favor of warnings.

beuerle, 2017-10-31:
I tend to agree with @arsenovic: as long as there is no conflict with another vi in skrf we should not care too much about the possibility and postpone a discussion of that to the future if/when it happens.
I worked with different kinds of instruments in the past, but I'm not aware of any conflicts in that regard (my memory might be shaky though, we hacked together our own _pyinstruments_ library then).

But at least for different instruments of the same type a check would be good to have. (I apologize for not having followed the scheme from the keysight YAML, a lot of commands I just hacked in from the docu without checking it properly; which is also the cause for some formatting issues @dvincentwest fixed).

As the number of people working on that is limited I think it would be enough to check for possible conflicts without enforcing any naming scheme.
At least as a starting point, we could let the parser read the template SCPI YAML file and then check the individual instrument files and report any conflicts. It would then be up to the developer to react to this and possibly fix the conflict and rerun the parser.

```
$ ./check_functions.py commands

conflicting function names:

                    keysight_pna         rs_zva
                    -------------------- -------------------- 
CALC:PAR:SEL        selected_meas        par_select
...
```

An additional check the parser can perform is the reverse case, if function names are implemented to perform different SCPI commands and report on that. In some cases this is necessary and correct, as different vendors sometimes put the same function into different sub-trees:

```
$ ./check_functions.py functions

functions with different branches: (or something like that)

                    keysight_pna         rs_zva
                    -------------------- -------------------- 
sample_function     SYS:DISP:SAM         DISP:SAM
...
```

cafeclimber, 2023-08-26:
This can be closed with the vi update

closed 2023-08-26 by jhillairet