scikit-rf/scikit-rf #186: NetworkSet handle different port numbers and frequency info

ldraughn, 2017-11-03:
At the moment it looks like the NetworkSet can only handle directories with files having the same number of ports and frequency information. Since individual network objects can be created with different port numbers and frequency info, can the NetworkSet not also be made to handle this difference.

arsenovic, 2017-11-05:
sure, but a lot of NetworkSet's functionality  would not work with heterogenous sets. what is the use case? why not just use a dictionary?

ldraughn, 2017-11-06:
A dictionary is how I am working around it now. The use case would be to plot and then reduce all the ports in a package model. These models typically get produced an grouped by channel and function so they don’t always have the same number of ports. This can be handled several other ways, but when using some this a Jupyter notebook to document your processing of the models and single line that reads all the SnP files a directory is much cleaner than 2-3 lines it take otherwise.

arsenovic, 2017-11-06:
i dont understand  your point. you can call me on google hangout to discuss if its easier.

dkozel, 2019-08-30:
I can't speak to the different port numbers, but one of the first things I tried to do with scikit was throw four s2p files into a `NetworkSet` to plot them and some of their attributes. I ended up using frequency slicing and resampling and just plotting things separately. I'll open as separate issue asking that the `resample` function of a `Network` allow start and stop frequencies to be specified so you get an exact set of frequencies out of the interpolation. Then you could dump everything into a `NetworkSet`.

jhillairet, 2020-03-01:
please reopen if you think the issue requires new development.

closed 2020-03-01 by jhillairet