scikit-rf/scikit-rf #193: Divide by zero error when trying to perform a SOLT calibration

kosmonout, 2018-02-02:
The generated error:

(base) C:\Users\XXXXXX\AnacondaProjects>python SOLT.py
C:\Users\XXXXXX\AppData\Local\Continuum\anaconda3\lib\site-packages\skrf\calib
ration\calibration.py:4456: RuntimeWarning: divide by zero encountered in true_d
ivide
  k_second =1/(Etr/(Erf + Edf *(Elr - Esf)))
C:\Users\XXXXXX\AppData\Local\Continuum\anaconda3\lib\site-packages\skrf\calib
ration\calibration.py:4456: RuntimeWarning: invalid value encountered in true_di
vide
  k_second =1/(Etr/(Erf + Edf *(Elr - Esf)))

I setup the calibration as follows:

import skrf as rf
from skrf.calibration import SOLT

measured_thrus = rf.Network('measured/thru.s2p')
measured_short = rf.Network('measured/short.s1p')
measured_shorts = rf.two_port_reflect(measured_short, measured_short)   
measured_open = rf.Network('measured/open.s1p')
measured_opens = rf.two_port_reflect(measured_open, measured_open)   
measured_load = rf.Network('measured/load.s1p')
measured_loads = rf.two_port_reflect(measured_load, measured_load)

ideal_thrus = rf.Network('ideal/thru.s2p')
ideal_short = rf.Network('ideal/short.s1p')
ideal_shorts = rf.two_port_reflect(ideal_short, ideal_short)   
ideal_open = rf.Network('ideal/open.s1p')
ideal_opens = rf.two_port_reflect(ideal_open, ideal_open)   
ideal_load = rf.Network('ideal/load.s1p')
ideal_loads = rf.two_port_reflect(ideal_load, ideal_load)
	
my_measured = [measured_thrus,measured_shorts,measured_opens,measured_loads]
my_ideals = [ideal_thrus,ideal_shorts,ideal_opens,ideal_loads]
## create a SOLT instance
cal = SOLT(
  ideals = my_ideals,
  measured = my_measured,
)
cal.run()
``

```

arsenovic, 2018-02-09:
i think the thru has to be last. try that.

arsenovic, 2018-02-09:
also, this might be answered faster on the mailing list, as its more of a use question not a code error (hopefully!)

kosmonout, 2018-02-09:
Thanks this works! But in the tutorial it is not explained correctly.

arsenovic, 2018-02-09:
Great , thanks for pointing that out

closed 2018-02-20 by arsenovic