scikit-rf/scikit-rf #196: Suggestion: Implement De-Embedding class

climbit, 2018-02-12:
Hi all,

is there intrest in implementing a class for de-embedding?

Cheers,
Daniel

denzchoe, 2018-02-13:
I've interest to attend it too :)

arsenovic, 2018-02-13:
i am not sure what you are envisioning. could you describe what you are
thinking, and  show some examples of how it would be used?

climbit, 2018-02-17:
The class should implement functions for different de-embeding methods like open pad de-embedding (opd), open short de-embedding (osd), 2 port de-embedding (2PD)

arsenovic, 2018-02-19:
ok, i am not familiar with those features. it sounds like partial one-port calibration tailored to on-wafer measurements?
you might be able to re-use some of the Calibration code, (like subclass Calibration), but if its easier for you to start fresh feel free to create a new file. you could put it in skrk/calibration/   but we can decide later.

climbit, 2018-02-21:
It is not necessarily tailored to on wafer measurements. For calibration you need very perfect standards because you assume a perfect short, open.... right? But usually you can't produce that perfect standards yourself. Therefore you need de-embedding, where you think of the parasitics as a network with different elements. You try to measure them and remove them from the dut measurements.

I open a new branch 'deembedding' in my fork and work within this branch. Should I push the progress from time to time or should I first push if I am done?

arsenovic, 2018-02-22:
for generic   calibration you can use any  standards, so long as you know what they are.  if you can model parasitics then you can use 'non-perfect' standards with normal SOL calibration.  alternatively you can use some self-calibration techniques. 

do you have an references or websites describing what you are disusing?

climbit, 2018-02-23:
> ... alternatively you can use some self-calibration techniques.

By self-calibration, you mean internal calibration of the equipment?

> for generic calibration you can use any standards, so long as you know what they are. if you can model parasitics then you can use 'non-perfect' standards with normal SOL calibration.

How do you know how a standard look like, don't you need to measure it?
How would the procedure look like with non-perfekt standards?
It would be great if you can show me some references. That might be very helpful for me.

Regarding what I said about calibration and perfect standards, I reference to [1], 2nd page, last paragraph.
[2] describes the methods I would like to implement.

[1] http://literature.cdn.keysight.com/litweb/pdf/5980-2784EN.pdf
[2] https://www.researchgate.net/publication/2517390_On-Wafer_Calibration_Techniques_for_Giga-Hertz_CMOS_Measurements

arsenovic, 2018-02-23:
self-calibration means you can solve for unknown parameters of the calibration standards at the same time you solve for the embedding circuit characteristics. TRL is the most popular 2port self-calibration. 

if you dont use self-calibration then you have to have a model for each standard, either simulated or guessed, see your keysight document  "The de-embedding process-the static approach'.

to me it looks like your second reference is describing some partial calibrations, which can be used if some assumptions are made on the error network (like symmetry, lossless, etc). this would  be useful, but would probably be best as implemented as a Calibration class.

we can talk over a google hang-out if you want more info.  

here is a book on calibration by a keysight guy, R&S has one too.  https://www.wiley.com/en-us/Handbook+of+Microwave+Component+Measurements%3A+with+Advanced+VNA+Techniques-p-9781119979555

chap 2 of my thesis has some talk about reflectometer calibration https://libraetd.lib.virginia.edu/public_view/qn59q4228
its not as good as the books, but its free!

climbit, 2018-03-01:
Hi Alex,

thank you for the offer. I would be glad to chat with you but unfortunately I do not have hang-out...
If you have signal, you could probably send me your number via email?

arsenovic, 2018-03-01:
Sure,. I think my cell is on my 810 lab website, call me

epallecchi, 2018-04-26:
Hi, 

I am also interested in a deembedding class, on wafer measurements here. I can do it with my own code, as it is essentially subtraction of Y/Z matrices, but it  would be nice to have it in skrf.

Can help testing, if needed.

arsenovic, 2018-06-25:
hi, 
    this issue seems to have petered out. any takes on implementing this?

vik-s, 2018-08-10:
Hi, 

I would like to contribute some code to deembedding classes for on-wafer applications. I have primarily been doing open/short deembedding. I suppose I can extend it to other types of deembedding too. What is the best way to approach this? Create a deembedding.py and put all the classes/methods there?

thanks!

arsenovic, 2018-08-14:
great ! 

to make sure we all understand this feature can we agree  any de-emedding  technique which is not a 'full' calibration is a partial calibration, or makes assumptions about the error networks?. in this case it sounds to me like sub-classing calibration would make the most sense. 

regardless, if   you would like to start fresh, then maybe you could put your code in skrf/calibration/onwafer.py?

vik-s, 2018-08-16:
I'm OK with treating de-embedding as a partial calibration. I took a look at subclassing calibration and it looks doable. I will likely go with creating an "OpenShort" class that would be used like:

from skrf.calibration import OpenShort
op = rf.Network('open.s2p')
sh = rf.Network('short.s2p')
dut_raw = rf.Network('dut.s2p')
measured = [op, sh]
dm = OpenShort(measured=measured)
true_dut = dm.apply_cal(dut_raw)

These methods are not restricted to on-wafer but can widely be used as long as the fixture to be removed can be considered small in terms of wavelength - where lumped approximations work. Due to the scale of devices on-wafer, this happens to be true on many occasions.

I'll look into this and submit a PR when it's ready.

arsenovic, 2018-08-16:
great, i think that is the best way.

jhillairet, 2020-07-26:
Eventually this should be documented somehow.

jhillairet, 2021-07-27:
closed with PR #480

closed 2021-07-27 by jhillairet