scikit-rf/scikit-rf #199: scikit-rf 0.14.6 broken

skortchmark9, 2018-02-22:
As of today, pip installs fail: https://pypi.python.org/pypi/scikit-rf/0.14.6

```
    running install_data
    error: can't copy 'LICENSE.txt': doesn't exist or not a regular file
```

arsenovic, 2018-02-22:
yeah i was trying to fix this, but i dont  think i can re-upload a new dist to pypi. try it now with what should be 0.14.7

skortchmark9, 2018-02-22:
Yes, works for me - thanks! Can you also remove the broken version from pypi? https://stackoverflow.com/questions/20403387/how-to-remove-a-package-from-pypi

arsenovic, 2018-02-22:
yeah i did that. i meant i cant re-upload the same version. so i had to bump to 14.7.  which isnt building on the conda-forge anyway so i might have to bump again. my release technique is bad.

skortchmark9, 2018-02-22:
Gotcha - thank you.

closed 2018-02-22 by skortchmark9