scikit-rf/scikit-rf #201: Cannot import scikit-rf 0.14.7

skortchmark9, 2018-03-01:
Hello again,

I was able to successfully install scikit-rf, but I could not import it.

```python
>>> import skrf
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File ".../venv/local/lib/python2.7/site-packages/skrf/__init__.py", line 17, in <module>
    from . import media
  File ".../venv/local/lib/python2.7/site-packages/skrf/media/__init__.py", line 50, in <module>
    from .freespace import Freespace
  File ".../venv/local/lib/python2.7/site-packages/skrf/media/freespace.py", line 23, in <module>
    from ..data import materials
  File ".../venv/local/lib/python2.7/site-packages/skrf/data/__init__.py", line 26, in <module>
    ntwk1 = Network(os.path.join(pwd, 'ntwk1.s2p'))
  File ".../venv/local/lib/python2.7/site-packages/skrf/network.py", line 378, in __init__
    fid = get_fid(file, 'rb')
  File ".../venv/local/lib/python2.7/site-packages/skrf/util.py", line 168, in get_fid
    return open(file, *args, **kwargs)
IOError: [Errno 2] No such file or directory: '.../venv/local/lib/python2.7/site-packages/skrf/data/ntwk1.s2p'
>>> 
```

arsenovic, 2018-03-01:
how did you install skrf?

skortchmark9, 2018-03-01:
```pip install scikit-rf```

arsenovic, 2018-03-01:
do you know what version that is ?

pip list |grep  scikit-rf
?

skortchmark9, 2018-03-01:
0.14.7

arsenovic, 2018-03-06:
try to update now, and let me know if it works. should be 0.14.17

arsenovic, 2018-03-06:
i mean 0.14.71

arsenovic, 2018-03-06:
actually try 0.14.8.  my bad

closed 2018-03-23 by arsenovic