scikit-rf/scikit-rf #207: Time domain plot is not working

BKHrf, 2018-03-22:
I have tried your cool package and it seems to work great so far,

I have, however, some problems plotting the time domain response of my S-parameters (See attachment).
It seems that not only I have the issue, since the code on the scikit-rf website also have some issues: 
http://scikit-rf.readthedocs.io/en/v0.14.8/examples/networktheory/Time%20Domain.html

[image]

arsenovic, 2018-03-22:
@mhuser ideas ?

i also noticed some of my time-gating  code is  working, but not producing the same results as it used to .

mhuser, 2018-03-23:
Hum. Latest version is ok, isn't it:
http://scikit-rf.readthedocs.io/en/latest/examples/networktheory/Time%20Domain.html

mhuser, 2018-03-23:
@BKHrf , do you use the latest version which is master on github?

BKHrf, 2018-03-23:
I just downloaded the package through Anaconda. It states that I use 0.14.8.

mhuser, 2018-03-23:
I am terribly sorry for this inconvenience. I understand now that we had time-domain broken in this version. There is two solution to this:
- Installing the latest version from github (way of the brave)
  `pip install git+https://github.com/scikit-rf/scikit-rf`
  (probably required to remove prior version with conda first)
- Convincing @arsenovic to bump the version on Pypi due to popular demand.

I would propose to do the second, as the time-domain may be interesting for many. What is your view @arsenovic ?

arsenovic, 2018-03-23:
yes, your advice is good @mhuser ,

 ideally i could replace 14.8, but pypi doesnt let me replace versions for integrity reasons. i had this problem recently and had to make a bunch of wasted releases, which take a long time to test through conda-forge. 

however, here is a demo of how the plot_s_db_time() has changed from 14.5 to 14.8(actually head on master)
https://gist.github.com/arsenovic/920c9f2879d9843b7677ecb092850359
so the current version is  wrong. i am trying to trouble shoot this.  once i fix this i can make a new release.

arsenovic, 2018-03-26:
fixed in https://github.com/scikit-rf/scikit-rf/pull/208, and subsequent prs

mhuser, 2018-03-27:
Great!
Works good with the new release v0.14.9.

BKHrf, 2018-03-29:
Looks great now. Thanks for the fix.

closed 2018-06-25 by arsenovic