scikit-rf/scikit-rf #213: Loading Simulation S data into SciKit-RF

GProtoZeroW, 2018-04-22:
I understand that Scikit-RF was originally written for use of reading and analyzing the data from a Real Network Analysers such as KeySight in the data format that those use; However, I am trying to use the functionally of your library for simulation work I am doing with PySpice. Where I have setup a test network to measure the wave values 
$a=\dfrac{V+Z_0I}{2\sqrt{Z_0}}$ ect 
and calculted my S values from there where the V, I values are from PySpice(NgSpice) ac sweep. Where each $S_{xy}$ is stored as a colloim in a pandas dataframe along with one collum for the frequency that where sweept. 

The issue I am having is after trying several times I have no idea how to load my dataframe of values into Scikit-RF which I would then use to calculate the other matrices, transfer functions, Smith Charts ect. 
So if you could help that would be much appreciated. 
I can post a link to the notebook with sim in a day or two. Thanks

dvincentwest, 2018-04-22:
Hi @GProtoZeroW, for future reference, there is a google group that is setup for community support questions like this:

https://groups.google.com/forum/#!forum/scikit-rf

Since we are hear already, you may as well take a look at the keysight_pna.py module which contains a get_snp_network function that is very similar to what you are looking at

https://github.com/scikit-rf/scikit-rf/blob/8700168f0df9db16ba3330fffe317b6f4ebfaaa6/skrf/vi/vna/keysight_pna.py#L216

The function gets a 1-D array from the instrument containing R/I S-parameter values which are processed into an array of complex values, and lastly are constructed into a network object.   See the whole function for more details but below is the snippet where we take the "data" array which contains both frequency and the real, imaginary values and add them into the network.  Note that for a network the ntwk.s property is the scattering matrix itself which is a complex numpy array with shape( len(frequency), nports, nports)

```python
fdata = data[0]
sdata = data[1:]
ntwk = skrf.Network()
ntwk.frequency = skrf.Frequency.from_f(fdata, unit="Hz")
ntwk.s = np.empty(shape=(sdata.shape[1], nports, nports), dtype=complex)
for n in range(nports):
    for m in range(nports):
        i = n * nports + m
        ntwk.s[:, m, n] = sdata[i * 2] + 1j * sdata[i * 2 + 1]
ntwk.frequency.unit = f_unit
ntwk.name = name
return ntwk
```

closed 2018-04-22 by dvincentwest

arsenovic, 2018-04-22:
Hi @GProtoZeroW, 

  i think your best bet is to post this question to the mailing alist and attach a copy of your notebook. 
if you can generate  complex s-parameters  then it should not be hard to convert dataframe to network

GProtoZeroW, 2018-04-22:
what mailing list?; and at the moment PySpice is not returning complex from NgSpice; but I can just typecast to complex with a dummy 0j for the moment