scikit-rf/scikit-rf #217: one-port cal documentation error

ScottHinton, 2018-06-21:
Three issues:
1. In the one-port cal code you have a reference to the page http://na.support.keysight.com/vnahelp/tip20.html which I found very helpful.  It derives the equations behind the one-port cal, but if I'm understanding it correctly it looks like you have an error in your OnePort class docstring.  You give the fancy LaTeX-formatted equations that are to be solved, but it appears that the m's and the i's are swapped.  For example, you wrote "e_{11}\mathbf{i_1m_1}-\Delta e\,\mathbf{m_1}+e_{00}=\mathbf{i_1}" -> e11*i1*m1 - delta_e*m1 + e00 = i1, however it should be e11*i1*m1 - delta_e*i1 + e00 = m1.  This is properly reflected in the math implemented in the class' code.

2. I was confused for a while with your code because of your notation.  You appear to follow the same a, b, c notation as found in the reference I mentioned above, except for one slight difference: they define "c = -e11" but you use c = e11 (no negative), and the resulting equations that you use adapt accordingly.  I think it would be helpful to add a comment in your code explaining that minor difference.

3. One final, minor point: in OnePort.run() you have a comment "# ASSERT: mList and aList are now kx1x1 matrices, where k in frequency".  Actually k is equal to the number of frequency points in one's measurement times 4 (the number of S-parameters in the 2-port measurement) since you performed a "measured.s.reshape" operation.  Like I said, it's minor, but when people are not experts in this field and are still trying to navigate (in my case it's less like navigating and more like stumbling) their way through it, minor errors can become confusing obstacles.

I realize that I could propose these changes by cloning the project, making the edits, then making a pull request or something like that, but since I'm not a git expert I'd rather not spend the time figuring out how to do it since that would take an order of magnitude more time than simply making the corrections.  Please forgive my laziness.

ScottHinton, 2018-06-21:
It looks like the interpreter changed some of my shorthand equation with asterisks into italics, *like this*.  Obviously I'm a markdown newbie.

arsenovic, 2018-06-25:
yes, you are right, thanks for pointing this out, we have been misleading the masses!

arsenovic, 2018-06-25:
scott, i fixed the docstring.
that whole function is really old and could probably be cleaned up a lot. you should make a git account do a PR just as an exercise. you will likely have to work with software and some point ....

im not sure about your 3rd point, the measured and ideals should be 1-ports.

ScottHinton, 2018-06-26:
It would be good practice, however by the time I actually need to use it for work I will surely have forgotten all the little details that I had to figure out in order to make it work.

You are right about #3 above, they are one-ports.  I remember that you referenced the OnePort cal in the SOLT cal, but I went back and looked and you are slicing the measured and ideal S-parameters into one-ports before calling OnePort.  So I was wrong on that one.

closed 2018-06-26 by ScottHinton