scikit-rf/scikit-rf #219: Wrong common line selection in Multiline TRL calibration

wanyancan, 2018-08-08:
Hi,
I found that in multiline TRL calibration, the procedure to find the best common line to use always gives 0 as the common line.

Line 2769 in calibration.py
            min_phi_eff = pi*npy.ones(lines)
            #Find the best common line to use
            for n in range(lines):
                for k in range(n-1):
                    dl = l[k] - l[n]
                    pd = abs(exp(-gamma_est*dl) - exp(gamma_est*dl))/2
                    if -1 <= pd <= 1:
                        phi_eff = npy.arcsin( pd )
                    else:
                        phi_eff = 0
                    min_phi_eff[n] = min(min_phi_eff[n], phi_eff)
            #Common line is selected to be one with the largest phase difference
            line_c[m] = npy.argmax(min_phi_eff)

arsenovic, 2018-10-13:
@Ttl , do you have any thoughts on this?

Ttl, 2018-10-13:
It seems to be true. Lines are compared to lines before it, which there are none for the first one and it always uses default value of pi that is maximum that arcsin can give.

Comparing to every line should match the "Multiline revealed" paper:
```diff
diff --git a/skrf/calibration/calibration.py b/skrf/calibration/calibration.py
index 62be871..7a7d592 100644
--- a/skrf/calibration/calibration.py
+++ b/skrf/calibration/calibration.py
@@ -2769,7 +2769,9 @@ class NISTMultilineTRL(EightTerm):
             min_phi_eff = pi*npy.ones(lines)
             #Find the best common line to use
             for n in range(lines):
-                for k in range(n-1):
+                for k in range(lines):
+                    if n == k:
+                        continue
                     dl = l[k] - l[n]
                     pd = abs(exp(-gamma_est*dl) - exp(gamma_est*dl))/2
                     if -1 <= pd <= 1:
```
With synthetic data it gives the same results, but with real data there are some small differences if there are multiple lines.

This is how a matched line looks like before and after:

Old:
[image]

Fixed:
[image]

There seems to be more discontinuities in S-parameters with the fixed common line selection although the results are probably more accurate since the matching is overall better. I'll test with some more data at work next week.

There is an improvement to multiline TRL calibration that avoids choosing a common line and instead does some magic with tensor decomposition (https://ieeexplore.ieee.org/document/7162897). It should avoid the discontinuities and be more accurate, but its pretty tricky to implement.

wanyancan, 2018-10-14:
Is there a tensor decomposition library with CPD method in python? @Ttl

Ttl, 2018-10-16:
I tested with some more real data and they had similar discontinuities. I think I prefer using the thru always as the common line. It isn't really any less wrong choice.

I'm not familiar with tensor decompositions.

wanyancan, 2018-10-17:
I agree. The discontinuities are not acceptable even with the improvement on averaged error.
I tested using TensorLab in Matlab, the error can be improved 5x to 10x smaller on average with no discontinuities. The main issue I have is the speed using the tensor decompositions.

closed 2018-11-26 by wanyancan