scikit-rf/scikit-rf #225: Mline Loss too high

arsenovic, 2018-10-10:
https://groups.google.com/forum/#!topic/scikit-rf/FKfn36C4zPE

@mhuser, any ideas about this?

mhuser, 2018-10-13:
`MLine` media is a microstripline model which implementation is strongly inspired from [Qucs](http://qucs.sourceforge.net/) one, which is based on various papers (see source
 code for references).

It has been shown that the S21 results from the skrf implementation divert from other tools, like AWR Microwave office.

[image]

After a quick comparison with Qucs, it happens that the shape of the S21 looks good, but the amplitude has a poor agreement.

Let's see what is happening there.

mhuser, 2018-10-13:
There was a extra `sqrt(er_eff)` factor in the microstripline `alpha_conductor` computation. The conductor loss is only influenced by surface resistance and skin effect and the geometrical dimensions have not to be corrected in relation with `er_eff`.
The implementation results are consistance with Qucs microstripline model implementation.

AWR and Qucs microstripline model seems to lead to different results, but that is another question. If we use another tool, we probably get a third result, depending of the microstripline model used.

[image]

PR #226 (2018-10-13): Fix MLine alpha_conductor computation
Fix #225 
There was a extra `sqrt(er_eff)` factor in the microstripline `alpha_conductor` computation. The conductor loss is only influenced by surface resistance and skin effect and the geometrical dimensions have not to be corrected in relation `with er_eff`.
The implementation results are consistance with Qucs microstripline model implementation.
closed 2018-10-13 by arsenovic