scikit-rf/scikit-rf #231: ComplexWarning: Casting complex values to real discards the imaginary part

alexatadi, 2018-11-17:
Got below complex warning:

> i:\sci\Anaconda3\lib\site-packages\skrf\media\coaxial.py:154: ComplexWarning: Casting complex values to real discards the imaginary part
  '\nCharacteristic Impedance=%.1f-%.1f Ohm'%(self.Z0[0],self.Z0[-1]) +\
i:\sci\Anaconda3\lib\site-packages\skrf\media\coaxial.py:155: ComplexWarning: Casting complex values to real discards the imaginary part
  '\nPort impedance Z0=%.1f-%.1f Ohm'%(self.z0[0],self.z0[-1])

To reproduce:

> from skrf import Frequency
from skrf.media import CPW, Coaxial
freq = Frequency(1,10,101,'ghz')
coax = Coaxial(frequency=freq, Dint=1e-3, Dout=2e-3)
coax

jhillairet, 2018-11-17:
Dear alexatadi,

Yes indeed. This is due to the fact the coaxial line characteristic
impedance has a very tiny imaginary part. This tiny imaginary part comes
from the conductor conductivity of the coaxial line, which is setup by
default to INF (infinity). However, INF is set to 1e99, which is high but
still lead to non-zero imaginary part.

I agree that the display may be confusing. Would it be better if it would
display instead:
"Characteristic Impedance=(41.6,-0.0j)-(41.6,-0.0j) Ohm"
?

Julien

jhillairet, 2018-11-18:
> Hi Julien,
>
> Yes I agree with this proposal, since we already set the INF to 1e-99,
> then follow ths clip to show results would be better. Thanks a lot!
>
> Regards,
> Alex

OK, I will make a PR to correct the warning display.

jhillairet, 2018-11-18:
PR merged on #232 to solve the issue.

closed 2018-11-25 by arsenovic