scikit-rf/scikit-rf #237: se2gmm: mixed-mode calculation for 4-ports seems not correct

alexatadi, 2018-11-24:
Algorithm comes from: https://www.rohde-schwarz.com/us/faq/calculate-the-mixed-mode-s-parameter-from-an-s4p-file-faq_78704-30300.html

I tried to calculate the gmm s-parameter with below 2 ways, showed aligned for 2-ports network, but obviously different results for 4-ports network:

1. origional se2gmm()
2. the way from R&S, by modified origional se2gmm() with:
```python
from numpy.linalg import multi_dot as npy_multi_dot

aa=npy.array([1, -1, 1, 1]).reshape(2, -1)
bb=npy.zeros([2, 2])
cc=npy.c_[npy.r_[aa, bb], npy.r_[bb, aa]]/npy.sqrt(2)
## cc=aa # for 2-port
for ii in range(self.s.shape[0]):
    self.s[ii]=npy_multi_dot([cc, self.s[ii], npy_inv(cc)])
```
Please let me know if who need more detailed results, the s4p file and 2 results, thanks.

arsenovic, 2018-11-25:
@core-dd, any ideas?

JAnderson419, 2019-07-15:
This may be a bit late to the show, but I have used se2gmm successfully in my own code in the past. I used the R&S page as a guideline, but had to reorder things before doing the transformation as the matrices are in slightly different forms. Here is a snippet from my own code (available in my PyMeasRF repo, sParmAnalysis.py) where I do the transformation, not including reading the 

```

switch23 = True  # Switches position of ports in code so that mixed mode input is 12
                # and output is 34. Default is 13 input, 24 output.
bal = False  # Set true if importing true mode SNP data

rfdata = open_Network(os.path.join(datadir, f))

            if bal:
                # raw data is mmdata in form 11 12 with each submatrix as dd dc
                #                            21 22                        cd cc
                for i, freq in enumerate(rfdata.f):
                    # reorder 4x4 matrix for each freq seperately
                    # to decrease memory usage
                    rfdata.s[i, :, :] = gmm_reorder(rfdata.s[i, :, :])
                sedata = cp.deepcopy(rfdata)
                sedata.gmm2se(2)
                sedata.renumber([0, 1, 2, 3], [0, 2, 1, 3])
                vals = [f, sedata, rfdata]
            else:
                if switch23:
                    rfdata.renumber([0, 1, 2, 3], [0, 2, 1, 3])
                mmdata = cp.deepcopy(rfdata)
                # renumber ports for single ended to differential conversion
                mmdata.renumber([0, 1, 2, 3], [0, 2, 1, 3])
                mmdata.se2gmm(2)
                # mmdata now in form  Sdd  Sdc
                #                     Scd  Scc
                vals = [f, rfdata, mmdata]
```

closed 2021-03-18 by arsenovic