scikit-rf/scikit-rf #238: Can I use a coaxial cal kit (N-type) that is expressed in terms of coefficients

DavidLutton, 2018-12-03:
Using coefficients from the [85032F Type N (50) Calibration Kit](http://na.support.keysight.com/pna/caldefs/PNA/85032F.htm)
Given that [TwelveTerm](https://scikit-rf.readthedocs.io/en/latest/api/calibration/generated/skrf.calibration.calibration.TwelveTerm.html#skrf.calibration.calibration.TwelveTerm) requires ideals, is there a way to generate the ideals for use in scikit-rf?

METAS [VNA Tools Math](https://www.metas.ch/dam/data/metas/Fachbereiche/Hochfrequenz/VNA_Tools/VnaToolsMath%20V2.0.pdf) in section 6.1, demonstrates the maths to make use of a Agilent/Keysight cal kit   
`# Implementation of (186) for the open standard` 
`f = 1e9`
`C0 = 89.939`
`C1 = 2536.800`
`C2 = -264.990`
`C3 = 13.400`
`Ceff = C0 + (C1*f) + (C2 * f**2) + (C3 * f**3)`  # Correct ?
`print(Ceff)`
1.3399999735010003e+28`

Would [tlineFunctions.reflection_coefficient_2_input_impedance_at_theta](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.tlineFunctions.reflection_coefficient_2_input_impedance_at_theta.html#skrf.tlineFunctions.reflection_coefficient_2_input_impedance_at_theta) play a part for the effect of the delay?

Thank you for creating this library, I had a quick read & play around today. 
I'm pleased how simple it was to open a .s2p file, as an example evaluate if a dut (a isolator) was a reciprocal network `max(dut.s21.s-mag - dut.s12.s-mag) > 10` and also plot it

arsenovic, 2018-12-06:
i cant see the metas link, but i think their definition  is a keysight model is a series of capacitors?
if so you can use the [media](https://scikit-rf.readthedocs.io/en/latest/tutorials/Media.html) class to generate the circuit.   if the model is made up of only capacitors, any media instance should work (since no gamma/z0 should be needed). 

```
from skrf.media import Freespace
frequency = sample_ntwk.frequency # or define frequency from scratch
med = Freespace(frequency=frequency, z0=50)
f = frequency.f
Ceff = C1*f + C2....
ideal_open = med.capacitor(eff)**med.open()

```

arsenovic, 2018-12-10:
does this work for you?

DavidLutton, 2018-12-14:
I have made some progress [here](https://gist.github.com/DavidLutton/c530da778bd4adc94e4b10b2d5c17611) 

Updated [METAS VNA Tools Math](https://www.metas.ch/dam/data/metas/Fachbereiche/Hochfrequenz/vna-tools/vnatoolsmath-v2-1-0.pdf) link

I don't know how to express the 'Loss (Gohm/Sec)', I think could this be [DistributedCircuit distributed resistance, in Ohm/m](https://scikit-rf.readthedocs.io/en/latest/api/media/generated/skrf.media.distributedCircuit.DistributedCircuit.html) at a normalization frequency [Keysight App note Page 6](http://literature.cdn.keysight.com/litweb/pdf/5989-4840EN.pdf) of 1GHz
How can I convert and add this loss effect?

Thank you

arsenovic, 2018-12-14:
i think you have the right idea. the content in the  keysight app note should work.

arsenovic, 2019-02-08:
i think we can close this?

closed 2019-02-10 by DavidLutton