scikit-rf/scikit-rf #239: Filter responses of a bandpass, bandstop, highpass, lowpass

DavidLutton, 2018-12-30:
I have access to a number of filters without details, having response profiling tools would be useful to process them.
[I have written a function to return the 3dB points of a bandpass filter](https://gist.github.com/DavidLutton/328bc003c7632c541c8f68d457d419ee)

Is this suitable concept for inclusion in this project?

In this example the data export did something odd, shows as a bandpass filter with gain 
[image]

DavidLutton, 2018-12-30:
For highpass, just changing the source data and plot title
[image]

For lowpass, just changing the source data and plot title
[image]

arsenovic, 2019-02-08:
david, 
   not sure why i didnt respond to this. i think this would make a great example notebook. 

to make it a function i think you would will need  more general parameters like  3db-> any db, dont hardcode mhz, etc. but i like your idea, and it works great.

dl1chb, 2019-03-06:
Also estimating the filter's order would be nice (which could lead to estimate the poles and zeros ;-)).

DavidLutton, 2019-04-06:
- [x] Work on this issue in JupterLab, which is easier to iterate/retry
- [x] Get access to a VNA and collect samples of different filters: Band Pass, Low Pass, High Pass 
- [x] Removed the handling frequency parts and changed to working with data points
- [x] Fix incorrect selection of regions that don't fulfil the intended criteria (may need tuning)
- [x] ndB point to ndB (arbitrary) and optional offset
- [ ] Update the gist and samples
- [ ] Tag regions with the criteria that they were identified by

`filter_region(net.s11, 0, -3, offset=np.max(net.s11.s_db))`
`filter_region(net.s11, -3, -60, offset=np.max(net.s11.s_db))`
[image]
`filter_region(net.s11, 0, -3, offset=np.max(net.s11.s_db))`
`filter_region(net.s11, -3, -90, offset=np.max(net.s11.s_db))` 
[image]

jhillairet, 2020-03-01:
Dear @DavidLutton, is your function been updated/completed?

jhillairet, 2020-07-26:
Closing because no news. You can reopen if necessary.

closed 2020-07-26 by jhillairet