scikit-rf/scikit-rf #250: Proposal: Extracting LRC-Lumped Component Models from S-Parameters

dl1chb, 2019-03-06:
When simulating transmission lines in cadence virtuoso, a proper method is using equivalent circuit models of transmission lines created from em-simulations. Despite of the well known (simple) RLC model, there are a few other models beeing more broadband (consisting of several inductors, resistors and capacities to proper model skin effekts and so on). They can be created in a size of about 1/20 of a wavelength and used in parametric simulations using cadence virtuoso.
Despite of extracting the effective permittivity from s-parameters, it'll be nice to extract those lumped elements as well.
Question is: what do you think about that? Has anyone done that yet? I am thinking about trying to implement this like in Media (lumped transmissionline for example). Is there an interest in that?

jhillairet, 2019-03-06:
Hello,

Un principle this is possible, but not automatic.

I've deduced the equivalent lumped model of a tunable capacitor, from a set
of full wave simulations of this capacitor made at different tuning
positions.
I've made an equivalent circuit with lumped components and I fitted its
component values with scipy.optimize.curve_fit to match both real and
imaginary parts of the impedance of the full wave networks. But this way I
need to assume in advance the circuit setup...

I guess many better and advanced procedures exist to perform this kind of
task...

Julien

dl1chb, 2019-03-06:
Hi,

what did you need to assume from the circuit setup? Du you mean the
circuit, in which the capacitor was placed or about how the circuit from
the capacitor will look like?

Beckus

-- 
I use GnuPG (GPG) for e-mail encryption and signing. If you want some privacy, my public key ID is 2F9D4F14.

Christopher Beck, M.Sc.
Akademischer Rat

Friedrich-Alexander-Universität Erlangen Nürnberg
Lehrstuhl für Technische Elektronik
Cauerstrasse 9 (Raum 3.28)
91058, Erlangen Deutschland

Tel. (+49)9131 / 85-27193
Web: http://lfte.de/~beck
Rufz: DL1CHB

jhillairet, 2019-03-06:
>
> what did you need to assume from the circuit setup? Du you mean the
> circuit, in which the capacitor was placed or about how the circuit from
> the capacitor will look like?
>
> I knew before that this real-life capacitor has a certain amount of self
inductance, so the equivalent circuit of this real-life capacitor is close
to an ideal RLC circuit.

Julien

dl1chb, 2019-03-06:
Ah, okay, in this case, I think we can life with that. There are some model topologies in literature, which could be implemented and also checked against each other (dependent on which kind of transmission line at a certain frequency and so on). In the case of a transmission line, it often is quiet clear, where which RLCG-Element hast to be. The models differ on their bandwidth, where they are applicable.

closed 2020-01-30 by jhillairet