scikit-rf/scikit-rf #251: Proposal: adding noise component to Network

cactorium, 2019-03-20:
Hello! Would there be any interest in adding noise information to `Network`? It looks like some s2p files already include some noise information along with the scattering parameters. This would be useful at least for me because I've been messing around with designing a transistor amplifier using this library. I'd be up for trying to code up some initial version of it, but I don't know exactly what you guys would like it to be designed. It seems like it'd involve a lot of mostly straight forward coding so that cascading networks works correctly, and it'd probably be nice if there were some way to calculate NF from it

arsenovic, 2019-03-20:
i think that is a great idea

TomWoo, 2019-03-21:
I'm very interested in this feature. Let me know if there's anything I can help with.

cactorium, 2019-03-21:
Awesome, and thanks! I'm not great at RF stuff so I'm definitely down for some help

Right now I'm actually confused about how to calculate the noise correlation matrix from the noise parameters. Qucs provides these formula: http://qucs.sourceforge.net/tech/node9.html , where I'm kind of stuck because I don't know where to get Tmin in c11, and Googling got me this one: https://www.cambridge.org/us/files/8313/6698/2373/HFIC_chapter_3_noise.pdf , where c11 is a lot simpler. Do you know which formula is better or more usable for this case? I'm converting to a correlation matrix because calculating network cascades is done using the correlation matrices instead of noise parameters, and I'm planning on using the ABCD matrix-based transform here: https://onlinelibrary.wiley.com/doi/pdf/10.1002/9781119073093.app3

There's also the question of how to store the noise data, since it uses an independent set of frequencies than the scattering parameters. Right now I'm thinking of just keeping them separate and interpolating during cascades or when .interpolate() is called. Does that seem like a sane design approach?

arsenovic, 2019-03-21:
@jstenar  have you experience with noise data?  do you have suggestions about a design architecture?

TomWoo, 2019-03-21:
The page from Qucs generalizes an n-port network, whereas the 'simpler' formulation is for a 2-port, which I think is all I would ever use. Storing a separate set of freq's makes sense to me, to minimize loss of numeric precision.

cactorium, 2019-03-22:
Got some code written, please review

FournierVincent, 2020-03-27:
Noise calculation of cascaded networks is of major interest. Congratulation for the work done.

It's **deembed** conterpart would be extremely useful as well. 
For information, it is implemented in the ADS **de_embed** component.

An immediate use-case will be noise modelling of a dut by removing it's access path contributions to losses **and** noise. The Noise deembedding method has a tremendous advantage over the traditional method based on losses only : The smithchart coverage of the tuner is much higher. 5x in my case. 

The reference IEEE articles below seem to be good guides for documentation and calculation implementation. 

The original IEEE article of Hillbrand on noise calculations gives a general approach of combining  series,parrallel, cascade (and associated Z,Y,ABCD representations) of the correlation matrix of two-ports networks. 

The IEEE article of Pucel gives another interesting feature by inverting the cascade equation : Noise deembedding !!!
This is of main interest for me. I would like to see that implemented in scikit-RF. 
For information, it is implemented in the ADS **de_embed** component.

The IEEE article of Rizzoli broadens the calculation to N-ports. 

The IEEE article of kellog applies to passive networks. 

references
[1] H. Hillbrand and P. Russer, "An efficient method for computer aided noise analysis of linear amplifier networks," in IEEE Transactions on Circuits and Systems, vol. 23, no. 4, pp. 235-238, April 1976.

[2] R. A. Pucel, W. Struble, R. Hallgren and U. L. Rohde, "A general noise de-embedding procedure for packaged two-port linear active devices," in IEEE Transactions on Microwave Theory and Techniques, vol. 40, no. 11, pp. 2013-2024, Nov. 1992.

[3] V. Rizzoli and A. Lipparini, "Computer-Aided Noise Analysis of Linear Multiport Networks of Arbitrary Topology," in IEEE Transactions on Microwave Theory and Techniques, vol. 33, no. 12, pp. 1507-1512, Dec. 1985.

[4] K. Kellogg, L. Dunleavy and A. D. Snider, "Temperature dependent noise system verification and the relationship of passive noise parameters to available gain calculations," 2018 IEEE 19th Wireless and Microwave Technology Conference (WAMICON), Sand Key, FL, 2018, pp. 1-4.

cactorium, 2020-03-27:
I'd be interested in working on implementing this if you guys would want it added to the library

jhillairet, 2020-03-27:
> 
> 
> I'd be interested in working on implementing this if you guys would want it added to the library

Sure, please feel free to propose something

FournierVincent, 2020-03-27:
The minimum Rn condition for positive noise contribution should be tested, followed by warning and correction if necessary. 

If NFmin, Rn, and Sopt, are used to characterize noise, the following relation must be satisfied for a realistic model:
[image]

where, T0 is the standard IEEE noise temperature (290 Kelvin) and 
T is the ambient temperature.
Since the reflection coefficient, S11, for the network in which Noisy2Port is embedded is not known, it is not possible to check. Instead, if the noise parameters NFmin, Rn, and Sopt describe a system that requires negative noise because Rn is too small, the negative part of the noise will be set to zero and a warning message will be issued.

FournierVincent, 2020-03-28:
The deembed equation should be very similar to the cascade. 
cascade : CA = A1 CA2 A1t  + CA1

deembed  : CA2 = A1.inv    (CA - CA1)   A1t.inv
(this is my guess, need to check in ref [3,4])

Since the cascade equation is in the pretty long connect() function,  I would think of introducing a "deembed" option to either
1-the connect() function 
2- a (new) network property. 
option 1 seems ugly to me. 
option 2 seems better but may impose restrictions in the usage of .inv() method.

There are restrictions on the ntwkA network :
It should not be "noisy". In that case force noise to null and generate a warning before proceeding with the calculation. 

 <partial code>
if NetwA.deembed :
      cC = npy.matmul(a, npy.matmul(  (cB - cA), a_H)) 
else :
      cC = npy.matmul(a, npy.matmul(cB, a_H)) + cA

call : 
NetwC = connect(NetwA.inv(), NetwB)

Reactions ?

If I try to contribute despite my low experience in github, should I fork from scikit-rf/scikit-rf or from cactorius/scikit-rf ?

cactorium, 2020-03-28:
@FournierVincent Thanks for all the useful comments! If you want to try to implement it, definitely do it off of the master branch of scikit-rf/scikit-rf My changes were merged in over a year ago, so it's much better to work off of the latest working version of it. I'm not very versed in RF stuff, where do you think the deembed function should go?

FournierVincent, 2020-03-28:
Hello Cactorium, 
There are various way of specifying noise for a component, Touchstone reading is one of them, but not the only one. 
ADS does allow the user to specify a two port network having S=[0.01, 2.3, 0.13, 0.01], NFmin=2.5dB, Rn=10, GamOPt=10+j35 at some freq.
I can't see any reason why scikit rf would not. 
Vincent

FournierVincent, 2020-03-28:
I made a first attempt. It seems to work as expected on a basic example. 
I updated the below repository 
https://github.com/FournierVincent/scikit-rf
Is it a proper way to use github ?

cactorium, 2020-03-29:
> Hello Cactorium,
> There are various way of specifying noise for a component, Touchstone reading is one of them, but not the only one.
> ADS does allow the user to specify a two port network having [...]

What I meant is that the current implementation only allows noise to be entered in two specific ways. Other ways of doing it simply aren't written and this aren't supported

cactorium, 2020-03-29:
> I made a first attempt. It seems to work as expected on a basic example.
> I updated the below repository
> https://github.com/FournierVincent/scikit-rf
> Is it a proper way to use github ?

Now you should open a pull request and work with the maintainers to get that change integrated in

FournierVincent, 2020-03-29:
**1/ remarks on noise calculation**
- property N.f_noise 
	should handle single point frequency. 
	Single frequency analysis is very common for noise  circles (contour plots)
- missing setter for parameter N.nfmin, N.rn, g_opt, ...
- missing rn_min_valid(nfmin, g_opt) function
- missing noise parameter checker 
N.rn should be checked against the rn_min_valid() function
- property N.noise =>  N.CA
	N.noise should be defined as  N.CA since it refers to the "a" representation of the correlation matrix. 
- missing property N.CY, N.CZ  
- missing operator CA->CY, CA-CZ, CY->CA, CZ->CA, CY->CZ, CZ->CY
	N.CA, N.CY, N.CZ representations and the translation operators are mandatory for a general noise calculation. 
- missing property C.T 
	The temperature of each component's instance needs to be defined independently
- function nf() :
	The nf() function should handle the component's instance temperature. 
**Nice to have :**
- missing setter for property N.CA, ...
- missing property : N.noise_bypass
The user should be able to make a component noiseless even if it is "noisy". 

**doc**
- 	documentation is missing
- 	example missing

**2/ integration of noise into skrf**
- missing skrf property : system temperature
	The system temperature is a key parameter for circuit noise simulation. 
- missing skrf property : noise_bypass
The user should be able to make a simulation noiseless even if it there are "noisy" components. 

- function renormalize() should call the nf() function

**doc** 
- 	Network.add_noise_polar and similar functions should be removed form the Noise chapter and transfered to a new "blurring" chapter (or whatever meaningfull name)
- 	side remark on function renormalize() : 
		The user's attention should be drawn to the fact that the renormalize function can be considered as a circuit simulation with matching circuits at port ends, giving S parameters and noise parameters.

FournierVincent, 2020-03-29:
I made above an extensive list of wishes on noise calculation. 

After some discussion, filtering and prioritizing, we should come up with something feasible allowing real improvements in the noise field. 

My immediate needs, which are key for noise deembedding within a mmwv source pull measurement setup are :
- setter for the noise parameters (g_opt, nfmin_db, rn) 
- single frequency operation.

FournierVincent, 2020-03-29:
> Now you should open a pull request and work with the maintainers to get that change integrated in
Isn't that too early ? I am not familiar with working in a group of developers. I think that  some additional improvements/verifications should take place before merging.  What is your opinion ?

TomWoo, 2020-03-29:
> I made above an extensive list of wishes on noise calculation.
>
> After some discussion, filtering and prioritizing, we should come up with something feasible allowing real improvements in the noise [...]

Both of these features should be straightforward to implement.

1. Try adding g_opt, nfmin_db, rn to the following list under the network class:
[image]
and define property setters for each of them.
It would also be a good idea to extract out noise correlation matrix creation from read_touchstone() into the noise property while you're at it.

2. The single-frequency case should already work when there's only one element in the f_noise array. If you're seeing a datatype issue (scalar instead of array), I'm sure there's also a quick fix.

My opinion is that correlation matrices shouldn't be directly mutable and should only be calculated from noise parameters (or potentially s-params for passive 2-ports). Property getters for CY and CZ can then be defined using z2y and a2z functions.

TomWoo, 2020-03-29:
> The deembed equation should be very similar to the cascade.
> cascade : CA = A1 CA2 A1t + CA1 (1)
> deembed : CA2 = A1.inv (CA ) A1t.inv (2)
> 
> NB : this is my guess, need to check in ref [3,4].
> CA2 = A1.inv (CA - CA1) A1t.inv from (1)
> NetwA is not noisy => CA1 = 0
> 
> Since the cascade equation is in the pretty long connect() function, I would think of introducing a "deembed" option to either
> 1-the connect() function via **kwargs
> 2- a (new) network property to be set by the **floordiv** operator , ie .inv()
> option 1 :
> option 2 : may impose restrictions in the usage of .inv() method.
> 
> There are restrictions on the ntwkA network :
> It should not be "noisy". In that case force noise to null and generate a warning before proceeding with the calculation.
> 
> comment ? Reactions ?
> 
> If I try to contribute despite my low experience in github, should I fork from scikit-rf/scikit-rf or from cactorius/scikit-rf ?
> 
> `
> 
> called by :
> NetwC = connect(NetwA.inv(), NetwB)
> 
> def connect(NetwA, NetwB) :
> ...
> if NetwA.deembed :
> inv_a = s2a(NetwA.inv()) **-1
> inv_a_H = conjuguate_transpose(inv_a)
> cC = npy.matmul( inv_a, npy.matmul( (cB - cA), inv_a_H ))
> 
> else :
> cC = npy.matmul( a , npy.matmul(cB, a_H)) + cA
> 
> `

Using inv() for de-embedding makes more sense if possible because it maintains feature parity for the operation on noise and s-params. Regarding the limitation, I don't think I understand why ntwkA needs to be noiseless. Is there a numeric singularity issue? How does ADS handle this?

TomWoo, 2020-03-29:
> **1/ remarks on noise calculation**
>
> * property N.f_noise
>   should handle single point frequency.
>   Single frequency analysis is very common for noise  circles (contour plots)
> * missing setter for [...]

Rather than go over the entire list, here are the low-hanging fruit to start with:
1. Add a temperature field in Kelvin (T_K?), and use it in CA (would noiseCA be more clear?), which is already used to compute noise figure.
2. Make T_K an optional argument in nf() and nf_db() functions.
3. Check against minimum rn
4. noise_bypass
I can help review/verify code if needed. It's generally a good idea to write at least one unit test (see test_network.py) per feature.

cactorium, 2020-03-29:
> Isn't that too early ? I am not familiar with working in a group of developers. I think that some additional improvements/verifications should take place before merging. What is your opinion ?
> 
> > Now you should open a pull request and work with the maintainers to get that change integrated in

Yeah it's probably not ready to merge in but opening a PR creates a place to discuss changes to get it in a state where it could be added

I'll leave it to you guys to discuss details, it looks like Vincent's going to do the implementation

FournierVincent, 2020-03-29:
> Vincent's going to do the implementation

I am not sure to have the energy to handle the all thing, considering my tiredness and the short duration of the medical certificate. 

> Regarding the limitation, I don't think I understand why ntwkA needs to be noiseless. Is there a numeric singularity issue? How does ADS handle this?

ADS stops on error. There were users complaining but ADS claims the behavior is rigorous. 

I couldn't find any evidence of this statement in ref[2] where the only restriction for the network to be deembeded  is to consist of linear passive elements and avoid an isolator-like behavior. , but can still be non-reciprocal. (btw, I can't understand the non-reciprocal statement in this context. )

I will change the noise deembedding implementation so that it reflects the full deembedding, and drop the noiseless condition for now.  It would be interesting to have access to the ADS's comments as well.

TomWoo, 2020-03-29:
Ok, do what you can. I may have time to work on some of the remaining items later in the week or next week.

Reciprocal just means the network s-parameter matrix is symmetric. You can check for reciprocity within a tolerance using is_reciprocal().

Gloamglozer, 2020-04-26:
Hi guys, just chiming in to say that i'd like to help out implementing this. Any project tracking going right now?

Wanted to post here too that [an inverse exists for the ABCD NCM](https://github.com/scikit-rf/scikit-rf/files/4534605/NCM_Inverse_Proof.pdf), so maybe a `Network.inv()` can be defined for noise params. 

Also, what do you guys think about keeping this development confined to a noise branch until this is a bit more mature? I've already had some of my code break after upgrading to 0.15 (which i'll post about in a separate issue).

jhillairet, 2020-04-26:
> Also, what do you guys think about keeping this development confined to a noise branch until this is a bit more mature? I've already had some of my code break after upgrading to 0.15 (which i'll post about in a separate issue).

Yes you are right. 

BTW what is the issue you run into? port order in cascading?

FournierVincent, 2020-05-09:
Hi Eric, 

There is a realease forecasted early next week. 

I tryed to implement your ideas but failed. 

I think your ideas are better than mines and would like to see them implemented. 

I wonder wether you received the proper information to start merging your ideas into scikit-rf ? 

On top of that, I am also interested to discuss some ideas about noise parameters measurements. There are no real specialists in my lab and I would like an external view on a simple idea. 

Vincent Fournier, 
RF board design, RF measurements, IOT applications, embedded SW 

+33 6 01 84 88 97 
vf.fournier@free.fr 
www.linkedin.com/in/vincentfournier 

----- Mail original -----

Gloamglozer, 2020-05-10:
@FournierVincent Hi again -- sorry I ended up getting busy with another project and kind of let this slip away.  I'll try to look over what you've done sometime soon and make a list of questions for you, as well as things that we can change to keep the API clean.  I likely won't be able to add what I'm thinking of doing before the release so maybe it's better to hold off on merging 'till we have some real confidence everything is working great and a nice test suite for the noise features. 

I'll get in touch on the Slack about a call

FournierVincent, 2020-05-18:
Hi Eric, 

I could not read your version since the link was not sent. Could you please send it back ? 

I am currently looking more deeply into the behavior of ADS version of the (noise) deembedding. 

and compaired to the current version of skrf (mine). 

Meas1 => P1-------------------------------- S2P(ckt.s2p) ----Amp(nfmin, gopt,rn) -----P2 
Meas2 => P1---- deembed(ckt.s2p) ------S2P(ckt.s2p) ----Amp(nfmin, gopt,rn)-----P2 

skrf : Meas2 retrieves the initial noise parameters (nfmin, gopt, rn) 
ADS : doesn't retrieve and complains about negative noise due to rn too small. 

When rn is augmented (*2) in meas2, nfmin, gopt can be retrieved which is fine. 

How does your version perform ? 
What is the good behavior ? 

I would be very gratefull if we could have a small call/chat at your convenience (day or night no matter) 

Vincent Fournier, 
RF board design, RF measurements, IOT applications, embedded SW 

06 01 84 88 97 
vf.fournier@free.fr 
www.linkedin.com/in/vincentfournier 

----- Mail original -----

jhillairet, 2020-07-26:
Closing as many things have been made by @FournierVincent . Please reopen if necessary.

closed 2020-07-26 by jhillairet