scikit-rf/scikit-rf #254: Proposal: Advanced Circuit

jhillairet, 2019-04-08:
Connecting many Networks together can be a bit burden presently. 

I've started writing a new class named for the moment `Circuit` which is constructed by passing a list of connection. I've implemented the method described in [1], which apply to any combination of N-ports. 

For instance, if one want to create the following Wilkinson divider circuit [1]:
[image]

the current implementation is:

```
connections = [
           [(port1, 0), (branch1, 0), (branch2, 0)],
           [(port2, 0), (branch1, 1), (resistor, 0)],
           [(port3, 0), (branch2, 1), (resistor, 1)]
        ]

C = rf.Circuit(connections)
```
where the `connections` is a list of all intersections of the circuit, describing for each which Networks and associated port are connected to the intersection.

Presently, there are few constraints:
- all Network must have a name (and distinct ones).
- The ports must be 1-port Network and the string "port" included the Network name

For example, in the example above, ports have been defined like :

```
# Ports must be defined as Network
def Port(frequency, name, Z0=50):
    '''
    Return a port Network. Passing the name is mendatory
    '''
    _media = rf.media.DefinedGammaZ0(frequency, Z0=Z0)
    return _media.match(name=name)

port1 = Port(freq, name='port1')
port2 = Port(freq, name='port2')
port3 = Port(freq, name='port3')
```

The other components are standard Networks, for example:
```
freq = rf.Frequency(start=1, stop=2, npoints=101)
Z0_ports = 50

# resistor
R = 100
line_resistor = rf.media.DefinedGammaZ0(frequency=freq, Z0=R)
resistor = line_resistor.resistor(100, name='resistor')

# branches
Z0_branches = np.sqrt(2)*Z0_ports
line_branches = rf.media.DefinedGammaZ0(frequency=freq, Z0=Z0_branches)
branch1 = line_branches.line(90, unit='deg', name='branch1')
branch2 = line_branches.line(90, unit='deg', name='branch2')
```

The class also provide a way to test if the circuit is correctly connected (all ports of all Networks should be connected) and to visualize the graph of the circuit : 

[image]

Once the circuit is created, one can get the associated Network, that is the Network corresponding to 'external' ports. In theory, we can also have access to internal port, that is to the scattered waves, not only at the external ports, but also in all connections internally in the network. 

Let me know if it could be part of scikit and if you have any other comments.

[1] P. Hallbjörner, Microw. Opt. Technol. Lett. 38, 99 (2003).

arsenovic, 2019-04-22:
i think this would be a great feature. while i dont think you can beat the usability of a spice-like gui for large circuits,   i do frequently have had to connect handfuls of networks and dont not want to leave skrf, so this would be great in that use case.

jhillairet, 2019-05-10:
Closed with PR #257

closed 2019-05-10 by jhillairet