scikit-rf/scikit-rf #255: stitch doesn't work with network2

vincent766, 2019-04-08:
stitch seems not to work with the new network2 format. 

The stitch method uses the _network.f_ feature. I think it should refer to _network.frequency.f_ instead.

Vincent

arsenovic, 2019-04-22:
network2 is just  a prototype, never completed.

closed 2019-04-22 by arsenovic

vincent766, 2019-04-22:
Hello Alex, 

I discovered your tool recently. I struggled with ADS linear simulator before. 

I found scikit-RF 100x better than ADS for all the non-standard calculation tasks. 
It works really well. There are no deadzones when combining two ideas for example. 

Thank you for the tool creation and maintenance. 

I will make some trials on my side for extending network2 and possibly submit it to github if it makes sense. 

Bye, 
Vincent 

----- Mail original -----