scikit-rf/scikit-rf #256: calibration coefficents

sixtemesseven, 2019-04-11:
Is it possible to extract the eg. SOL calibration values from a set of S-Parameter measurments using the scikit-rf calibration functionality?

So that I could input the ideal and the measured s2p of the SOL set and get coefficents required by a VNA back?

Like: 
https://www.kirkbymicrowave.co.uk/Support/FAQ/What-are-the-calibration-kit-coefficients/

Sorry if this was asked before!

arsenovic, 2019-04-11:
yes. there are a couple methods  for this , here is one.
https://scikit-rf.readthedocs.io/en/latest/api/calibration/generated/skrf.calibration.calibration.Calibration.coefs_3term_ntwks.html#skrf.calibration.calibration.Calibration.coefs_3term_ntwks

sixtemesseven, 2019-04-11:
> yes. there are a couple methods for this , here is one. https://scikit-rf.readthedocs.io/en/latest/api/calibration/generated/skrf.calibration.calibration.Calibration.coefs_3term_ntwks.html#skrf.calibration.calibration.Calibration.coefs_3term_ntwks
> […](#)

Nice, thank you :)

closed 2019-04-11 by sixtemesseven