scikit-rf/scikit-rf #260: apply_cal cannot be applied to NetworkSet

cweickhmann, 2019-05-03:
Is there a reason apply_cal (for instance in skrf.calibration.TRL) cannot be applied to a NetworkSet?

MWE:
```python
import skrf as rf

trl = rf.TRL([thru, refl, line])

net = rf.Network("somefile.s2p")
net_set = rf.NetworkSet.from_dir("./some_measurements", name="Some Measurement")

cal_net = trl.apply_cal(net)   # <-- this works
cal_net_set = trl.apply_cal(net_set)   # <-- this fails

# Work around:
cal_net_set = [trl.apply_cal(n) for n in net_set]
```

The problem with this work around is that a property like "name" in this case is not saved.

jhillairet, 2019-05-04:
I don't think there is a specific reason it could not be applied. This is a good idea.

arsenovic, 2019-05-04:
yes this would be welcome. there are methods like `apply_cal_to_list`, and `apply_cal_to_all_in_dir`

cweickhmann, 2019-05-05:
We often have the case that a bunch of measurements is in a NetworkSet for work-flow reasons.
@arsenovic, you're the first to point this out. Am I blind or are those functions not documented? (It's probably the first ...)
If I have time, I'll try to write an implementation.

arsenovic, 2019-05-06:
https://scikit-rf.readthedocs.io/en/latest/api/calibration/generated/skrf.calibration.calibration.Calibration.html#skrf.calibration.calibration.Calibration

arsenovic, 2019-05-06:
i think you can just write a little  method for your use, something like ,

`my_cal.apply_cal_to_ns = lambda x: rf.NS(my_cal.apply_cal_to_list(x))`

if you want to use it in the future or have stable scripts then its better to add it to skrf.

cweickhmann, 2019-05-07:
Thanks, @arsenovic!
As mentioned above, the problem I have with going via a list of Networks is losing the name attribute.
Fork is underway ;-)

closed 2019-05-13 by arsenovic