scikit-rf/scikit-rf #263: GPIB communication fails using a Keysight E5071C VNA

gaussoreuler, 2019-05-10:
I'm trying to communicate with a Keysight VNA using a scikit virtual VNA, however I am getting an error when creating skrf.vi.vna.PNA object. The GPIB address seems to be right. I am suspecting that the scikit-rf version and the pyVISA i have installed  aren't compatible with each other. I get the following error message using scikit version 0.14.9 and pyVISA version 1.9.1:

Traceback (most recent call last):
  File "Documents\Python_scikit-rf\get_switch_data.py", line 10, in <module>
    vna = skrf.vi.vna.PNA(address='GPIB0')
  File "C:\Users\Edelmann\AppData\Local\Programs\Python\Python37-32\lib\site-packages\skrf\vi\vna\keysight_pna.py", line 36, in __init__
    super(PNA, self).__init__(address, **kwargs)
  File "C:\Users\Edelmann\AppData\Local\Programs\Python\Python37-32\lib\site-packages\skrf\vi\vna\abcvna.py", line 109, in __init__
    self.write = self.resource.write
AttributeError: 'GPIBInterface' object has no attribute 'write'

arsenovic, 2019-05-13:
@dvincentwest  any thoughts?

arsenovic, 2019-05-13:
is that GPIB address correct? the default looks like "GPIB::16::INSTR"?

beuerle, 2019-05-13:
> is that GPIB address correct? the default looks like "GPIB::16::INSTR"?

I think the `::INSTR` is optional, but the double `::` between the bus name `GPIB` and the primary address is necessary.

: [VISA resource names](https://pyvisa.readthedocs.io/en/stable/names.html)

| Interface | Syntax |
|--|--|
| GPIB INSTR | `GPIB[board]::primary address[::secondary address][::INSTR]` |

gaussoreuler, 2019-05-13:
Alright I did not know that  ``::`` was necessary. The address I have used was ``GPIB0``. Now that I am using ``GPIB::0`` I get the following error message:

File "Documents\Python_scikit-rf\get_switch_data.py", line 11, in <module>
    vna = skrf.vi.vna.PNA(address='GPIB::0')
  File "C:\Users\Edelmann\AppData\Local\Programs\Python\Python37-32\lib\site-packages\skrf\vi\vna\keysight_pna.py", line 40, in __init__
    self.use_ascii()
  File "C:\Users\Edelmann\AppData\Local\Programs\Python\Python37-32\lib\site-packages\skrf\vi\vna\keysight_pna.py", line 49, in use_ascii
    self.resource.write(':FORM:DATA ASCII')
  File "C:\Users\Edelmann\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 223, in write
    count = self.write_raw(message.encode(enco))
  File "C:\Users\Edelmann\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 201, in write_raw
    return self.visalib.write(self.session, message)
  File "C:\Users\Edelmann\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1866, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))
  File "C:\Users\Edelmann\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_INV_SETUP (-1073807302): Unable to start operation because setup is invalid (usually due to attributes being set to an inconsistent state).

My code is:
```
import skrf
import skrf.vi.vna

vna = skrf.vi.vna.PNA(address='GPIB::0')
```

arsenovic, 2019-05-13:
maybe try to connect and communicate with just visa , not skrf. see what happens.

beuerle, 2019-05-13:
You can also try to use [PyVISA-py](https://pyvisa-py.readthedocs.io/en/latest/) as the VISA backend, to rule out problems with the NI VISA installation:

```python
import skrf
import skrf.vi.vna

vna = skrf.vi.vna.PNA(address='GPIB::0', visa_library='@py')
```

Additionally, although it is obvious: are you sure that no other instrument in your GPIB setup is using the same address? If you're not already doing that, connect just the VNA.

gaussoreuler, 2019-05-13:
I have tried using pyvisa to get the name of the address but this code does not give me any output:
```
import visa

rm = visa.ResourceManager()
rm.list_resources()
```

I get the sames error message when I use this code instead of using skrf:
```
import visa

inst = rm.open_resource('GPIB::0')
print(inst.query("*IDN?"))
```

jhillairet, 2020-09-25:
seems there is a link with #393