scikit-rf/scikit-rf #267: DC Extrapolation Not Working as Intended

denzchoe, 2019-05-14:
Hi All,

I am noticing that the DC extrapolation does not make any difference for different `kind` of interpolation type. I have tried ‘linear’, ‘nearest’, ‘zero’, ‘slinear’, ‘quadratic’, ‘cubic’. Non of it works.

Original Example is from [Read the Docs](https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/Time%20domain%20reflectometry,%20measurement%20vs%20simulation.html)

Proof:
```python
import skrf as rf
import numpy as np
from numpy import real, log, log10, sum, absolute, pi, sqrt
import matplotlib.pyplot as plt
from scipy.interpolate import interp1d
rf.stylely()
rf_plot = rf.plotting

file_path = 'Measurements/20190508/SL'

meas = rf.read_all_networks(file_path,contains='P_SL')
meas_dc_ext = meas.copy()
meas0_NwkSet = rf.NetworkSet(meas)   

#extrapolation type
ckind = 'linear'

for k,n in enumerate(meas):         #The enumerate() function adds a counter to an iterable.
    meas_dc_ext[n] = meas_dc_ext[n].extrapolate_to_dc(points = 4351,kind = ckind)

t3 = meas_dc_ext['K2_S3_P_SL'].s[:,0,0]

#recreate Nwk Set for comparison w/o extrapolation
meas_dc_ext_NwkSet = rf.NetworkSet(meas_dc_ext)    

# =============================================================================
# #self written extrapolation
# =============================================================================
Oldfreq = meas0_NwkSet[2].f
Newfreq = meas_dc_ext_NwkSet[0].f

old_s11 = meas0_NwkSet[2].s[:,0,0]
intp_func = interp1d(Oldfreq,old_s11,kind = ckind, fill_value='extrapolate') 
new_s11 = intp_func(Newfreq)
#DC value must have zero imaginary part
new_s11[0] = new_s11[0].real
new_s11 = 20 * log10(new_s11)   #convert into dB

# =============================================================================
# #plot to see differences
# =============================================================================
plt.figure('Freq')
plt.title('S11_dB')
meas_dc_ext_NwkSet[2].s11.plot_s_db(linewidth = 3, label = 'scikit-rf-0.14.8')
meas0_NwkSet[2].s11.plot_s_db(label = 'original', linewidth = 2)
plt.plot(Newfreq, new_s11, label = ckind)
plt.xlim(0, 0.1e9)
plt.ylim(-50, -10)
plt.legend(fontsize = 5)
```

For '**Linear**'
[image]

For '**Cubic**'
[image]

I have [fork and updated the Network.py file](https://github.com/denzchoe/scikit-rf/blob/master/skrf/network.py) to fix this issue. 

[image]

[image]

But it is not coded elegantly **AND** it is not suitable to use with extrapolation type, `kind` = `rational`.

Are you guys seeing this issue too?

mhuser, 2019-05-18:
Hi @denzchoe,
This is the intended behaviour. Interpolation kind is not linked with the extrapolation of the DC point.
Interpolation is performed on the non-dc frequency points to turn them into an harmonic sweep with each frequency being a multiple of the frequency start. For this, the interpolation kind is used. Best results are usually achieved when no interpolation is required and the measurement is already a harmonic sweep (which could result in a lot of points if ones need a low frequency start and a high bandwidth).
The DC point is always extrapolated with a linear projection based on the two first frequency points in phase and magnitude form.
This is indeed very limited and won't work well in many case, for this reason, you can specify your own dc point that you might obtain using your prefered extrapolation computation or a-priori knowledge of the device under test. Example for a perfect 2-port thru:
`meas_dc = meas.extrapolate_to_dc(dc_sparam = [[0,1],[1,0]])`
More discussion [here](https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!topic/scikit-rf/cXgGYILckF4)

mhuser, 2019-05-18:
This is the reason interpolation kind has no influence on extrapolation.
That being said, it could be interesting to have a wider choice of extrapolation methods.
However, extrapolation of the DC point is a tricky thing that might lead to strange results, while most users might be used to get a nice results from their VNA TDR function "out-of-the-box"
I am not sure what is best to do here.

mhuser, 2019-05-18:
One issue with using `kind=...` is that `scipy.interpolate.interp1d` give
`ValueError: A value in x_new is below the interpolation range.`
when `fill_value='extrapolate` and `kind='quadratic'` or `kind='cubic'`.
If you have found a proper way to workaround this, kind could be used as the extrapolation method and issue a warning when `kind='rational'` with a fallback on linear extrapolation. 
Alternatively an additional extrapolation_kind parameter could be added, which would create room for other algorithm to be implemented in the future.

closed 2020-03-01 by jhillairet