scikit-rf/scikit-rf #272: Thoughts on allowing skrf.read_all to accept a file list

iancleary, 2019-06-14:
Is there other interest in modifying the read_all function to also accept a file list. https://github.com/scikit-rf/scikit-rf/blob/994b92b03602a2cdadf647d01d48603467128173/skrf/io/general.py#L262-L267 

There are situations in which I have a list of files that span multiple directories, and have already built a list of files from s2p or ntwk files.

It would be my desire to modify the function to allow a `files` parameter while keeping the same behavior if a dir is passed in (as per the current functionality)

## From: 
`def read_all(dir='.', contains = None, f_unit = None, obj_type=None):`

...
~~~~
out={}
for filename in os.listdir(dir):
        if contains is not None and contains not in filename:
            continue
        fullname = os.path.join(dir,filename)
        keyname = os.path.splitext(filename)[0]
~~~~

--------------

## To: 

`def read_all(dir='.', contains = None, f_unit = None, obj_type=None, files=None):`

~~~~
    out={}
    
    filelist = files
    if files == None:
        filelist = os.listdir(dir)
    
    for filename in filelist:
        if contains is not None and contains not in filename:
            continue
        if files == None:
            fullname = os.path.join(dir,filename)
            keyname = os.path.splitext(filename)[0]
        else:
            fullname = filename
            keyname = os.path.splitext(os.path.basename(filename))[0]
~~~~

--------------
The following code would be unchanged and the documentation would be updated.

Thoughts?

Thank you for your time and I would update the tests and documentation accordingly.

arsenovic, 2019-06-18:
i think this would be fine. i like that its backward compatible.   go ahead and make a PR.

iancleary, 2019-08-11:
@arsenovic thanks for the merge, would you please close this

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2020-03-01 by jhillairet