scikit-rf/scikit-rf #273: Req support "virual istruments" SDR

Red-Owl, 2019-06-18:
hi,
SDR have a rx and tx simultaneus. example limesdr, bladerf adalmpluto.
the frequency depend of the hardware 30mhz -3800, 70-6000 ....

arsenovic, 2019-06-18:
this is incoherent, please clarify .

Red-Owl, 2019-06-19:
ok
https://limemicro.com/products/boards/limesdr/ (30mhz - 3.8ghz)
https://www.analog.com/en/design-center/evaluation-hardware-and-software/evaluation-boards-kits/adalm-pluto.html#eb-overview  ( RF coverage from 325 MHz to 3.8 GHz Up to 20 MHz of instantaneous bandwidth Flexible rate, 12-bit ADC and DAC)
https://www.nuand.com/bladerf-2-0-micro/ 
http://www.ettus.com/all-products/UB210-KIT/ ....

helgrind, 2019-06-19:
I presume the suggestion is to support software defined radios as virtual instruments.

From my understanding of the current state of the project, all scikit-rf virtual instruments supported are communicated to using SCPI protocol, which is a standard instrumentation protocol supported by lab instruments and implemented over GPIB, USB or ethernet. Instrument manufacturers put in the effort to write device drivers that let their instruments use this very standardised protocol, which makes it easier to write software which uses them.

In contrast, most SDRs use a buffered stream to transfer data to the host, which would require some extra software to record that data and process it before scikit-rf could import it and use it. Essentially the SDRs are like one part of a full instrument (the front-end), so in order to "support" them this project would need a complicated additional app to make an SDR act like a VNA, plus any extra hardware needed for that to happen as well. Your best bet is to try and find another project that has built a VNA out of an SDR (or two!) and write a script to import data from that project. Perhaps if there is such a project then writing an SCPI driver, or a virtual instrument driver for scikit-rf, might not be so difficult.

So in summary, an SDR alone isn't a compatible instrument to use with scikit-rf, but could be built into one, but that would be a different project...

My two cents anyway!

Red-Owl, 2019-06-20:
many sdr use https://github.com/pothosware/SoapySDR/wiki you have a time :) to implement a SCPI <-> SoapySDR protocol interface ? the sdr in the list is full duplex (hackrfone is half duplex)
thank you for your time :))

helgrind, 2019-06-20:
Okay, so you can use one of the full duplex SDRs as a one-port vna, or with enough channels a two-port, but you also need directional couplers and potentially a switch, unless you use a second tx channel for port 2 source. Again, the SDR is just a couple of parts of a whole vna - supporting then would require writing an SCPI driver which also contains logic for driving the vna, or a separate vna app for the driver to interface with. You would then need to provide options for all the different configurations you could have the receivers in, etc. It's a big undertaking and I think it deserves its own project - an open source vna built with soapysdr drivers.

jhillairet, 2020-03-01:
Please reopen if necessary

closed 2020-03-01 by jhillairet